MarlinFirmware/Marlin #9452: [Bugfix 2.0.x]  TMC2130

psavva, 2018-02-03:
It seems we have introduced a bug again into the TMC2130 SPI implementation.
@teemuatlut @thinkyhead @Bob-the-Kuhn 

I have just updated to the latest Marlin, with the TMC2130 SW SPI Support, and it's broken again.

I am now using an unmodified REARM Pins file, which has the new CS pins on the Ethernet Header.
PINS P1_01, P1_04, P1_10, P1_14 for X, Y, Z, E respectively.

Please note that the redefines for MSBFIRST are popping up everywhere during compilation.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1691765/Configuration.zip)

```
In file included from Marlin\src\HAL/SPI.h:83:0,
from .piolibdeps\TMC2130Stepper_ID1493\src\source\TMC2130Stepper.cpp:3:
Marlin\src\HAL/HAL_LPC1768/LPC_SPI.h:26:0: warning: "MSBFIRST" redefined

#define MSBFIRST 0
 
In file included from .piolibdeps\TMC2130Stepper_ID1493\src/TMC2130Stepper.h:4:0,
from .piolibdeps\TMC2130Stepper_ID1493\src\source\TMC2130Stepper.cpp:1:
Marlin\src\HAL\HAL_LPC1768\include/Arduino.h:39:0: note: this is the location of the previous definition
#define MSBFIRST     1
```

LichtiMC, 2018-02-04:
... and I try to get it to work for about 24 hours now ...
FYI, if there is only one axis driven by a tmc2130-driver, it works...

psavva, 2018-02-04:
Are you using Ramps/Mega?  Or the re-arm/ramps combo?

Some changes got pulled in a few hours ago, which I think fixes the issue I am having...

Are you using TMC2130 Drivers,  if so, you need to enable  HAVE_TMC2130 and not HAVE_TMCDRIVER, could you upload both your config files, and I'll validate.

tcm0116, 2018-02-04:
This should now be fixed. Can you try the current bugfix-2.0.x branch?

LichtiMC, 2018-02-04:
Still doesn't work for me.
M906 returns correct current values for the steppers but no movement at all.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1692741/Marlin.zip)

psavva, 2018-02-04:
I can confirm that @tmc0116 PR has fixed the issue.
Closing this for the Re-ARM.

closed 2018-02-04 by psavva

thinkyhead, 2018-02-04:
@LichtiMC Have you been able to rule out a hardware issue? It sounds like something weird is happening there. Try swapping stepper drivers around, if possible.

moracabanas, 2018-11-09:
I am compiling The lastest Marlin 2.0.x bugfix for LPC1768 Re-Arm with TMC2130. Also enabled TMC debug  and I found it only works if I enable #define TMC_USE_SW_SPI. So HW SPI is not working for me following this scheme:

[image]

labotecno, 2018-11-09:
Try to check your pin and find the good One for your board , lpc rearm pin.h something like that

tcm0116, 2018-11-10:
@ moracabanas - if I remember correctly, the LPC1768 only has two hardware SPI interfaces, and they're used by the two SD cards (onboard and external) on the ReArm. I believe `TMC_USE_SW_SPI` is required for the ReArm board. @p3p does that sound right?

p3p, 2018-11-10:
Indeed with the Re-ARM Hardware SPI is only available on: 
```
  #define SCK_PIN           P0_15
  #define MISO_PIN          P0_17
  #define MOSI_PIN          P0_18
```
The other channel is used for the onboard sd card and on the Re-ARM headers you can't access it. 

If you use a display/controller on that SPI port it can't currently be used in Hardware mode as it is used by by both the display and sdcard. Probably best to just use Software SPI.

thinkyhead, 2018-11-11:
Worth adding a sanity check for this? Seems like it would prevent a lot of headaches.

moracabanas, 2018-11-11:
> Indeed with the Re-ARM Hardware SPI is only available on:
>
> ```
>   #define SCK_PIN           P0_15
>   #define MISO_PIN          P0_17
>   #define MOSI_PIN          P0_18
> ```
> The other channel is used for [...]

IBut I don't use any display and I have display support disabled on marlin so, is there any way I could use HW SPI on this board? I chose a 32bit board for my corexy printer because the speed I could achieve with my 0.9deg/step motors. If I am forced to use SW SPI it wont handle +30mm/s speed consistently. So, what are my options? Is there any stable board with motor driver sockets and HW SPI for my TMC2130? Thanks for your reply

moracabanas, 2018-11-11:
So I found your Re-ARM pins on my board due to this pinout diagram

[image]

I will try to modify SPI pins definition in order to get this working without LCD.

moracabanas, 2018-11-11:
I have read about SW SPI speed limit for about 30mm/s for TMC2130 but I think the driver still use Step Dir to work and SPI only for special commands. So is there any drawback about using SW SPI TMC2130? Speed is important for me as my printer is about 300x300x400.

p3p, 2018-11-11:
I don't know of a reason why the SPI frequency would limit print speed, (as you said you still use the normal step/dir signals), or any other limitation, once the config it sent it doesn't need updated .. definitely not in realtime.

Although as you are not using the hardware channel for anything that should also work for you anyway.

moracabanas, 2018-11-12:
Thanks you so much. I was worried about SPI in general on this drivers because Linear_Advance is not working with TMC2130 Extruder and I cannot find why it just stalls instead of extrude if that option is enabled.

thinkyhead, 2018-11-12:
Software SPI should be fine. Linear Advance stepper stalling is an entirely different issue, and most likely has to do with conflicting settings. (_e.g._, Linear Advance requires a very limited range of Jerk values to work properly.) There are several threads on Linear Advance that you may search for, and perhaps one of them will throw light on your issue.

yellobello, 2019-01-08:
Hi, marlineers!

Could it be that software SPI is corrupt again?
I just tried to build a re-arm with TMC2130s and LCD and I get a TMC CONNECTION ERROR message on the LCD. I used the pins on the ethernet connector for SPI as suggested above, and when I connect the scope to the soft SPI pins I do not even get a clock, which should be visible if soft SPI worked, right?