MarlinFirmware/Marlin #9454: very slow retraction/restart in dual extrusion mode

Quenta-and-Bees, 2018-02-03:
- Description: ---
In dual extruder mode, extremely slow (like 0.5mm filament/sec) retraction and restart is sometimes observed. The print can resume, though with too much thermal contact between nozzle / work, & ooze during retraction, blobs occur on the work, sometimes fatally ruining the print.

I doubted the Gcode first, but it seems like they make sense (the problematic retraction seems no different from normal retraction).

- Behaviour:---
The symptom can be seen amid a retract or the following restart; for instance, when you want to retract 10mm for tool change, 7mm can be retracted normally, and the remaining 3mm extremely slow. The same is for retract. It even looks like a pause, but the from the geared cold end I could observe very slow movement of the motor. Notice that it happens only when 2 extruders are enabled (the machine is working perfectly fine with single extrusion mode, I don't know whether it'll happen with 3 or more). It also happens only on some retract / restart (maybe about 5% of all the retracts/restarts in the whole print).

- Steps to reproduce:
  - Printer type: Delta   
  - Board: MKS Gen 1.4 (Remix of Arduino mega2560 / RAMPS1.4)
  - Extruder: E3D chimera (2-in 2-out)
  - Marlin version: 1.1.8
  - Microsteps: 1/32
  - Slicer / host: Simplify 3D
  - Printed from SD or USB cable
  - used with tool change retraction or with tool change script (occurs on both situations)

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1691712/configs.zip)

Squid116, 2018-02-04:
Could this be related to: #9250?

I see similar on normal retraction.

Quenta-and-Bees, 2018-02-04:
It looks quite similar, but is your case  #9250 also with dual extruders?

thinkyhead, 2018-02-17:
You both have 32x microstepping. So it does insinuate a potential value overflow. The `Planner::_buffer_segment` method would be the place to add debug output to see what's going on at that level.

Be sure to download the latest bugfix code for testing.

Squid116, 2018-02-19:
@thinkyhead I'll have to get back to this later - I have a backlog of thing to print, so now I have it working I'm going to avoid breaking it again for a while!

I am concerned about moving to a re-arm with SD-6128 drivers (128 microsteps) - will this impact other axes? particularly if it is something like a uint8_t overflowing.

thinkyhead, 2018-02-25:
> I am concerned about moving to a re-arm with SD-6128 drivers (128 microsteps) - will this impact other axes? particularly if it is something like a uint8_t overflowing.

All I can say is try and see. We'd love to hear how it goes. Baby steps.

boelle, 2019-02-20:
@Quenta-and-Bees did you test with the latest `bugfix-2.0.x`?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Quenta-and-Bees, 2019-03-06:
sorry, I just tested latest firmware; it seems the problem is solved.

boelle, 2019-03-06:
then click close below

but dont use 1.1.9, upgrade to 2.0

closed 2019-03-06 by Quenta-and-Bees