MarlinFirmware/Marlin #9456: PID tuning heatbed fails immediately

schorsch3000, 2018-02-03:
### Bug Report
- Description: PID tuning heatbed fails immediately 
- Expected behaviour: Running someting like `M303 C8 E-1 S50 U` will PID tune the headbed
- Actual behaviour: 
```
Send: M303 C8 E-1 S50 U
Recv: PID Autotune start
Recv:  T:48.00 /0.00 B:27.81 /0.00 @:0 B@:127
Recv: Error:Heating failed, system stopped! Heater_ID: bed
Changing monitoring state from 'Operational' to 'Error: Heating failed, system stopped! Heater_ID: bed
'
Recv: Error:Printer halted. kill() called!
```
This happens in way less then a second, this is not the thermal runaway protection (at least not as it should work),  this fails even if i try to tune below the current bed temperature.
- Steps to reproduce:
Just run the gcode provided in "Expected behaviour"

I'm not sure if i accidentally deleted the part that should ask me about my version, or if it doesn't exist.
Im running 1.1.8 from  0640bd028215a01ff9a01a46d4dcb859ba208a20. Everything else works fine (for weeks).

If there is anything that i can do to help, feel free to ask.
[marlin-pid-bed-fail-report.zip](https://github.com/MarlinFirmware/Marlin/files/1692057/marlin-pid-bed-fail-report.zip)

GMagician, 2018-02-03:
There was a fix on autotune. try bugfix.1.1 branch

schorsch3000, 2018-02-03:
That seems to work, thanks.
i think i don't get the branch-schema, is it documented somewhere?

closed 2018-02-03 by schorsch3000

GMagician, 2018-02-03:
Don't know exactly.
1.1 is stable release
bugfix 1.1.x is latest version, with fix and new bugs ;-)
bugfix 2.0.x is work in progress version 2 (for new 32 bits boards and old boards as well)

schorsch3000, 2018-02-03:
so, i'm staying on bugfix-1.1.x, since i like new bugs better then the old
ones :D

GMagician, 2018-02-03:
I'm on 2.0.x even with mega2560 and ramp 1.4 and it seems to works.
2.0.x will suppress 1.1.x in the near future

thinkyhead, 2018-02-05:
> so, i'm staying on bugfix-1.1.x, since i like new bugs better than the old ones :D

Truly. I'd almost release it as 1.1.9, but there's this annoying bug #9360 that needs a solution first. I think that if that were fixed, there'd be nothing to restrain the release.