MarlinFirmware/Marlin #9457: Neopixel compiling error and a question about EVENT_LEDS brightness

Seri-, 2018-02-03:
Compile Error

When I compile with  #define NEOPIXEL_STARTUP_TEST enabled i get the following error:

sketch\neopixel.cpp: In function 'void setup_neopixel()':

neopixel.cpp:48: error: 'safe_delay' was not declared in this scope

     safe_delay(1000);

                    ^

exit status 1
'safe_delay' was not declared in this scope

Not a huge thing as it is not really a important function but an error is an error.

As an additional question:
Is there a way to dimm the #define PRINTER_EVENT_LEDS function to some value below max output or to change the colors? I have found a lot of color settings in the leds.h and some in the neopixel.h but i don't know where i should start. 
The  #define NEOPIXEL_BRIGHTNESS 127 setting don't seem to change much as the white light while printing seems to be as bright as  "R255 U255 B255".

OperatorB, 2018-02-12:
#9033 (same, older post, follow that will you)

thinkyhead, 2018-02-20:
Duplicate of #9033
Solved.

closed 2018-02-20 by thinkyhead