MarlinFirmware/Marlin #9458: BUG/TYPO? Z_DUAL_ENDSTOPS uses XMAX instead of ZMAX

fabtopia, 2018-02-03:
Current code in Configuration_adv.h of bugfix-1.1.x and bugfix-2.0.x
//#define Z_DUAL_STEPPER_DRIVERS
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  //#define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    **#define Z2_USE_ENDSTOP _XMAX_**
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif

In my opinion this should be _ZMAX_

thinkyhead, 2018-02-17:
The choice is rather arbitrary, but x-max was chosen because it's usually the next available free plug, and quite often the probe is connected to the next free Z plug.

fabtopia, 2018-02-17:
Oke, but in line off the other two the zmax would be logical. But that's oke. There is thought about it and this means it's not a typo.

closed 2018-02-17 by fabtopia