MarlinFirmware/Marlin #9459: White screen using Viki2 display V1.1.8 and shifted display

Iamturbo1978, 2018-02-03:
### Bug Report
- Description: ---

Viki2 only showing a white screen after upgrading from V1.1.4 to V1.1.8. Everything else works.

I did a little poking around in "Conditionals_LCD.h", I tried changing the contrast to 35, but it did not work. I then replaced this code

`    #if ENABLED(miniVIKI)`
`      #define LCD_CONTRAST_MIN      75`
`      #define LCD_CONTRAST_MAX     115`
`      #define DEFAULT_LCD_CONTRAST  95`
`      #define U8GLIB_ST7565_64128N`
`    #elif ENABLED(VIKI2)`
`      #define LCD_CONTRAST_MIN       0`
`      #define LCD_CONTRAST_MAX     255`
`      #define DEFAULT_LCD_CONTRAST 35`
`      #define U8GLIB_ST7565_64128N`
`    #elif ENABLED(ELB_FULL_GRAPHIC_CONTROLLER)`
`      #define LCD_CONTRAST_MIN      90`
`      #define LCD_CONTRAST_MAX     130`
`      #define DEFAULT_LCD_CONTRAST 110`
`      #define U8GLIB_LM6059_AF`
`      #define SD_DETECT_INVERTED`
`    #endif`

with this code from my working V1.1.4

`    #if ENABLED(miniVIKI)`
`      #define LCD_CONTRAST_MIN  75`
`      #define LCD_CONTRAST_MAX 115`
`      #define DEFAULT_LCD_CONTRAST 95`
`    #elif ENABLED(VIKI2)`
`      #define DEFAULT_LCD_CONTRAST 35`
`    #elif ENABLED(ELB_FULL_GRAPHIC_CONTROLLER)`
`      #define LCD_CONTRAST_MIN  90`
`      #define LCD_CONTRAST_MAX 130`
`      #define DEFAULT_LCD_CONTRAST 110`
`      #define U8GLIB_LM6059_AF`
`      #define SD_DETECT_INVERTED`
`    #endif`

and the display came to life. I am a noob when it comes to programming, I understand the basics but that is about it. I don't know why my change worked, but it did. 

Also, the display is shifted to the right by 4 columns. Those 4 columns are still showing the splash screen no matter what is on the screen. On the main info screen they are there, on the menu screen they are there. I have no idea where to look to correct this issue.

Thanks for the help

Bob-the-Kuhn, 2018-02-04:
Please ZIP up Coniguration.h, Configuration_adv.h and pins_YOUR_BOARD.h files and drop it into your reply.

Iamturbo1978, 2018-02-04:
I am so sorry about that. Here are the files you requested.

[Marlin V1.1.8 files.zip](https://github.com/MarlinFirmware/Marlin/files/1693264/Marlin.V1.1.8.files.zip)

Bob-the-Kuhn, 2018-02-06:
I just put my VIKI2 on a 2560 + RAMPS using bugfix-1.1.x and the display is working correctly.

Please try bugfix-1.1.x

Bob-the-Kuhn, 2018-02-06:
I just saw an issue where the VIKI2 display freezes if SDSUPPORT is enabled.

I'm also seeing this.  I'll let you know when this is cleared up.

Iamturbo1978, 2018-02-06:
Sounds like I plan, I will try it tonight. Thank you

Iamturbo1978, 2018-02-07:
Ok, I think it has to be a problem on my side. I tried the current BugFix, and I still get the white screen. Almost like the contrast is turned up to high. I tried it with SDSUPPORT enabled and disabled. 

I also came accross a different problem with the current BugFix. With USE_CONTROLLER_FAN enabled, I get a compiler error of 

`C:\Users\******\AppData\Local\Temp\arduino_build_475351\sketch\Marlin_main.cpp: In function 'void controllerFan()':`
`Marlin_main.cpp:13086: error: 'controllerFanSpeed' was not declared in this scope`
` controllerFanSpeed = speed;`

But with USE_CONTROLLER_FAN disabled it compiles fine.
 
This is the first time I have seen this problem. I have included my whole Marlin BugFix files. Maybe there is something I am doing wrong. 

Thank you for all the help
[iamturbo1978_BugFix_02.06.2018.zip](https://github.com/MarlinFirmware/Marlin/files/1701521/iamturbo1978_BugFix_02.06.2018.zip)

Bob-the-Kuhn, 2018-02-08:
VIKI2 is broken on both the current bugfix-1.1.x and the current bugfix-2.0.x.

I did find a bugfix-1.1.x from May where the VIKI2 worked.

Hopefully tomorrow I'll have enough time to get to root cause on this.

Iamturbo1978, 2018-02-08:
Thank you so much for taking the time for this. 

I was using V1.1.4 for a while, and I didn't have any problems with my VIKI2 working. It worked first shot. Maybe that could give you another starting point.

I did notice with V1.1.8, the menu would not timeout. If I went to the move menu for example, it will stay there untill I back out and manually go to the main screen.

thinkyhead, 2018-02-14:
If we know the commit or general time-frame when the display was broken, #9394 also needs some love.

Bob-the-Kuhn, 2018-02-15:
Here's a version of **ultralcd_st7565_u8glib_VIKI.h** that should take care of the problems.  Please test and let me know how it works.

[ultralcd_st7565_u8glib_VIKI.zip](https://github.com/MarlinFirmware/Marlin/files/1726457/ultralcd_st7565_u8glib_VIKI.zip)

This is the code in PR# #9642.

Iamturbo1978, 2018-02-15:
Thank you so much for the help. I will try it tonight when I get off of work

Tailslide, 2018-02-16:
FWIW I am also having the same problem and this .h file fixed it for me, but now my screen is rotated.  I had to comment out this line in pins_Rambo.h : #define LCD_SCREEN_ROT_180

Iamturbo1978, 2018-02-17:
@Bob-the-Kuhn 

I tried the file you provided. I replaced it in my current V1.1.8. The display is still shifted to the right. I tried the current BugFix only once right now (have to get some prints done). When I did try it, the screen was still white, like the contrast is cranked all the way up. Here are a couple pictures of what I am seeing with the display shifted. Please excuse the fur in the screen... The joys of owning a husky.

[image]
[image]

Bob-the-Kuhn, 2018-02-17:
Huskies are neat dogs.  The neighbors had one & I really enjoyed it.

Long shot -  does the shift remain if you power cycle the LCD?  Sometimes LCDs get stuck in weird modes.

If 1.1.4 doesn't have the shift and the new file does then I'll need to investigate further.

Iamturbo1978, 2018-02-17:
They are great dogs, but the fur gets everywhere. It looks like we gain a dog every time we brush her with the amount of fur that comes off her.

Yes, the shift will stay there. The only thing that changes is the artifacts in the 4 column shift. Sometimes it looks like the pictures, and sometimes it would be random pixels on. 

1.1.4 did not have any problems when I used it. I can only imagine the headache this is causing. I'm sorry about that. I wish I knew more about C++ to help out.

Bob-the-Kuhn, 2018-02-18:
This isn't a headache.  We're more than glad to squash bugs & improve things.

I found a second init sequence for the ST7565 displays.  This one shifts my VIKI2 display 3 or 4 pixels and requires a much lower LCD contrast setting (M250 C24).

Try the attached and see if it makes your display happy.
[ultralcd_st7565_u8glib_VIKI.zip](https://github.com/MarlinFirmware/Marlin/files/1734379/ultralcd_st7565_u8glib_VIKI.zip)

Iamturbo1978, 2018-02-18:
I will try it out today, hopefully this squashes the bug. 

Thank you for everything

Iamturbo1978, 2018-02-18:
OK, using BugFix-1.1.x from the one I got yesterday. I replaced the VIKI file, and there is no white screen. It seams like the contrast is working correctly. Sadly, the display is still shifted to the right.

Iamturbo1978, 2018-02-18:
I did a little poking around the code and looking at the ST7565 datasheet. I noticed the display is being set to scan down (83h - 00h) instead of scan up (00h - 83h). So I thought, this is a 128 pixel wide display, and the display controller can control up to a 131 pixel wide display. Maybe the display is being drawn from right to left instead of left to right.

I tried changing the scan direction, but that did not change anything. So I changed the column start address from 10h to 14h. With a normal scan direction that should have moved the display to the right 4 additional columns, but it didn't. It moved it to the left 4 columns. So now the display is centered where it should be. This might just be a band-aid solution, or it could be a total fix. Here is the file I changed to get it working.
[ultralcd_st7565_u8glib_VIKI.zip](https://github.com/MarlinFirmware/Marlin/files/1734947/ultralcd_st7565_u8glib_VIKI.zip)

Iamturbo1978, 2018-02-18:
Unfortunately I don't know, and right now I can't pull it apart to check the board.  What I can say is I bought it from MatterHackers about 6 months or so ago. So I would assume it is a true VIKI2, and not a knockoff. 

I think I may have explained what I did wrong. The column address is split into (2) 4-bit nibbles, a high nibble, and a low nibble. I first changed the high nibble from x10 to x14, and my display did not like that. I changed it back, and then I changed the low nibble from x00 to x04. That is what corrected my display. The display would see the value of x14, but as x10 for the high, and x04 as the low. Hope that make sense.

Bob-the-Kuhn, 2018-02-19:
I changed the low nibble to 0x04 in both places & that shifted my entire display left by four pixels.  Not good for my display.  Glad to hear it made your display fully functional.

boelle, 2019-02-20:
@Iamturbo1978 is this still an issue with latest bugfix 2.0?

Iamturbo1978, 2019-02-20:
I'm sorry, I do not know. I have upgraded to a Duet WiFi.

closed 2019-07-06 by shitcreek