MarlinFirmware/Marlin #9464: [Bug][bugfix_2.0.x] Enhanced M114 does not compile

dembach, 2018-02-03:
### Bug Report
- Description: no compilation possible if #define M114_DETAIL in config-adv
- Expected behaviour: sucessful compilation ;-)
- Actual behaviour: Error: 
```
in file included from sketch/src/gcode/host/../../inc/MarlinConfig.h:41:0,
                 from sketch/src/gcode/host/M114.cpp:23:
sketch/src/gcode/host/M114.cpp: In function 'void report_xyze(const float*, uint8_t, uint8_t)':
sketch/src/gcode/host/M114.cpp:37:56: error: 'dtostrf' was not declared in this scope
       SERIAL_PROTOCOL(dtostrf(pos[i], 8, precision, str));
                                                        ^
sketch/src/gcode/host/../../inc/../core/serial.h:156:55: note: in definition of macro 'SERIAL_PROTOCOL'
   #define SERIAL_PROTOCOL(x)          MYSERIAL0.print(x)
```
- Steps to reproduce:
  - enable `#define M114_DETAIL` in configuration_adv.h
  - try to complie 

source used : bugfix_2.0.x as of 03.feb. 2018

GMagician, 2018-02-04:
Compiled now correctly (today merge).
Please post your configuration

dembach, 2018-02-04:
i will redownload and test.... gimme some minutes....

dembach, 2018-02-04:
still not working.... same error.....
[pins_RAMPS_FD_V1.txt](https://github.com/MarlinFirmware/Marlin/files/1693173/pins_RAMPS_FD_V1.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1693174/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1693175/Configuration_adv.txt)

Bob-the-Kuhn, 2018-02-04:
See PR #9481 for the root cause of this issue and the solution.

If Travis is happy then I'll merge the changes later today.

dembach, 2018-02-04:
thanks @Bob-the-Kuhn ... i will close the doublette.....

closed 2018-02-04 by dembach