MarlinFirmware/Marlin #9465: Solution for inductive sensor not working on MKS SBase 1.3 z_min sensor pin

tofifee, 2018-02-03:
Easiest solution so far for me:

A 2N222A transistor.

Connect z_min ground to collector, z_min sensor to emitter and probe sensor wire to base.

Advantage imo is that it works for the MKS SBase 1.3 - maybe other versions as well - and the clear binary logic being sensed by z_min when the probe triggers.

Hth

Bob-the-Kuhn, 2018-02-04:
99% chance that a diode will do the same thing.  If you remove the collector connection I expect it'll continue to work.

Another option is to use 5V inductive sensors.  NO and NC versions are available.
https://www.aliexpress.com/store/product/M18-8mm-sensing-DC-5V-NPN-NC-LJ18A3-8-Z-AX-5V-cylinder-inductive-proximity-sensor/1162433_32851947449.html
https://www.aliexpress.com/store/product/M18-8mm-sensing-DC-5V-NPN-NO-LJ18A3-8-Z-BX-5V-cylinder-inductive-proximity-sensor/1162433_32719316657.html?spm=2114.12010108.1000023.15.1674489efHoIim

tofifee, 2018-02-04:
The MKS Sbase 1.3 has a problem to use anything but physical endstops. From my testing it only senses when Z_Min-Sensor gets connected to Z_Min-Ground and not any logic change on Z_Min-Sensor pin alone and also not the probe pulling Z_Min-Sensor to ground. At least in my case.

Why not another sensor? Had that one and the transistor. Also stubborn: "There had to be a way". And I do like transistors since I got my first eletronics kit as a kid ;)

The board seems to react faster to the probe this way.
 
@reverse mode:
Late in the night, just cobbled together. Was a "why didn't I do that before" moment.

Bob-the-Kuhn, 2018-02-04:
Using a general purpose NPN transistor to buffer the signal is a good idea.  I like that better than using a FET because the NPN is less susceptible to ESD damage from handling.

Running the NPN in standard mode (emitter to ground, collector to Z_MIN input) is also acceptable.

Some users may need to add a series resistor between the base and the probe.  There's a possibility that the probe can pump too much current into the base and blow the transistor.

tofifee, 2018-02-06:
Concerning the series resistor:

One has to measure the probes resistance. To get to 5V sometimes only a single 7.1k/6.8k is needed. Depends in the probe naturaly.

houseofbugs, 2018-02-13:
The best way is to use an optocoupler to isolate the 12V from the board 5V.

tofifee, 2018-02-13:
Also yes

closed 2018-02-23 by thinkyhead