MarlinFirmware/Marlin #9471: Sensorless homing with TMC 2130 not working when homing all axes at once

Tamonir, 2018-02-04:
Hi
- Description:
I have some issues using sensorless homing feature of Fysetc 2130. SPI - resistor is soldered out, CF4 and CFG5 are properly soldered. When Homing via GCODE "G28 X", X homes successfully by sensorless homing and detects the end of the axis. Same for "G28 Y". However, if I say "G28" or "G28 XY", both axes are raming into the end without detecting anything.

- Expected behaviour:
Proper detection of sensorless X and Y endstops

- Actual behaviour:
"G28 X" and "G28 Y" working fine with sensorless homing but "G28 XY" and "G28" do not (ramming without detection)
- Steps to reproduce:
  - MKS Gen L 1.0
  - all axes TMC 2130 (so far properly working)
  - Marlin Tarantula (1.1.8 with quick setup from here ( https://github.com/JimBrown/MarlinTarantula )
  - My current config:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1692639/Configuration.zip)

Thanks in advance

lemmingDev, 2018-02-10:
//#define QUICK_HOME

make sure QUICK_HOME is disabled

Tamonir, 2018-02-11:
Thanks a lot! I would never have found that one. Works perfect!

thinkyhead, 2018-02-19:
See if this change to `gcode_G28` helps, and you may then be able to use `QUICK_HOME` again…
```diff
-     if (home_all || (homeX && homeY)) quick_home_xy();
+     if (home_all || (homeX && homeY)) {
+       quick_home_xy();
+       #if ENABLED(SENSORLESS_HOMING)
+         safe_delay(500); // Short delay needed to settle
+       #endif
+     }
```

commit e3370d09 (2018-02-19): Add a SENSORLESS_HOMING delay to quick_home_xy

Fix #9471
commit d4e4fe40 (2018-02-19): Add SENSORLESS_HOMING delay to quick_home_xy

Fix #9471
closed 2018-02-19 by thinkyhead