MarlinFirmware/Marlin #9473: [2.0.x] Bug? enabling ATX power supply causes machine to not function

gavatron3000, 2018-02-04:
### Bug Report

latest Bugfix as of today setup for Re-ARM corexz based machined
- Description: --- enabling an ATX power supply with default state off causes machine to not follow commands from host, disconnects and wont move

- Expected behaviour: --- machine follows commands from host after homing, and prints

- Actual behaviour: --- machine boots to a "STOPPED" state on the lcd, im able to do a G28 as well as a G29 P1 P3 etc but will not follow movement commands before or after this from a host. when a simple jog from octoprint is request machine disconnects from host and shows "STOPPED" on lcd again

- Steps to reproduce: the config eg below:
#define POWER_SUPPLY 1

#if POWER_SUPPLY > 0
  // Enable this option to leave the PSU off at startup.
  // Power to steppers and heaters will need to be turned on with M80.
  #define PS_DEFAULT_OFF
#endif

My machine works otherwise with this disabled and prints well

thinkyhead, 2018-02-17:
@MarlinFirmware/32bit-maintainers — Any thoughts what might be going on here? ReArm `PS_ON_PIN` is this:
```cpp
#define PS_ON_PIN         P2_12 // (12)
```
…and the `setup_power_hold()` function is called early in `setup()`.

p3p, 2018-02-17:
This is how my printer has always been configured will have a play but I've never encountered this problem.

gavatron3000, 2018-02-17:
FWIW since this issue for stability I've checked the "ignore any unhandled errors from firmware" in octoprint  as otherwise it won't print now. Haven't tracked that one down yet

thinkyhead, 2018-02-28:
Try this:
- Configure the machine with `PS_DEFAULT_OFF` disabled.
- Intentionally unplug the ATX power supply.
- Start the machine.
- Wait…
- Power on the ATX power supply.

gavatron3000, 2018-03-02:
ive removed my ramps ecosystem from my printer now and gone with the Duet electronics, so i wont be able test this anymore. ill close the issue as it seemed to be only affecting me. cheers fellas

closed 2018-03-02 by gavatron3000