MarlinFirmware/Marlin #9475: [FR] Sensorless(tmc2130) homing on Deltas (and other bots that home to max on Z)

Hywelmartin, 2018-02-04:
Today there is only two axis that is configurable from within Marlin (X and Y). If not TMC_ADV is used.

in * Configuration_adv.h it looks like this under // @Section TMC2130, TMC2208

//#define SENSORLESS_HOMING // TMC2130 only

#if ENABLED(SENSORLESS_HOMING)
#define X_HOMING_SENSITIVITY 8
#define Y_HOMING_SENSITIVITY 8
#endif

To safely implement this feature it could  be present ONLY on the preconfigured delta  files or checked during sanity check, that #define Z_HOME_DIR is -1(max)

In the future:

#if ENABLED(SENSORLESS_HOMING)
#define X_HOMING_SENSITIVITY 8
#define Y_HOMING_SENSITIVITY 8
//#define Z_HOMING_SENSITIVITY 8 // WARNING Use only with deltas/ bots that home to max
#endif

tcm0116, 2018-02-04:
I just got my Delta machine setup with TMC2130s, and this is on my list of things to work on.

Hywelmartin, 2018-02-04:
Obviously, me too, so i can test

dembach, 2018-02-04:
looking forward to that..... maybe more accurate than microswitches and  less cables ;-)

tcm0116, 2018-02-07:
If you're running 2.0.x, I just submitted #9516 which adds Z axis sensorless homing and enabled it to work on delta machines. I was going to wait until I got some testing on it before making a 1.1.x version, but I can if needed.

Hywelmartin, 2018-02-07:
I'm running 1.1.x(8) right now and don't have a 32 Bit board

tcm0116, 2018-02-07:
You can run 2.0.x on your 8-bit board. You'll just need to get the current
beta version of Arduino to be able to compile.

tcm0116, 2018-02-08:
I just pushed #9532, which are the changes applied to the bugfix-1.1.x branch.

Hywelmartin, 2018-02-09:
I take a look at it during the weekend

tcm0116, 2018-02-09:
It's been merged into the bugfix1.1.x and bugfix-2.0.x branches, so you can
just download whichever one you use and go from there.

Hywelmartin, 2018-02-09:
I meant... I test it in the wild....

Hywelmartin, 2018-02-13:
tested
it homes one axis at the time... not good
if I pull the sd card the printer reboots
if I try set delta hight the printer reboots
if I xxxxxx the printer reboots

closed 2018-02-20 by thinkyhead