MarlinFirmware/Marlin #9476: [2.0.x] STM32HAL compile errors with Arduino 1.8.5 and STM32duino boootloader

youprint, 2018-02-04:
I've a compilation error with the STM32F103RE config. I tried both linux or windows and i have the same error with spi. 

`sketch\src\HAL\HAL_STM32F1\HAL_spi_Stm32f1.cpp: In function 'void spiBeginTransaction(uint32_t, uint8_t, uint8_t)':
sketch\src\HAL\HAL_STM32F1\HAL_spi_Stm32f1.cpp:169:55: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'BitOrder' [-fpermissive]
   spiConfig = SPISettings(spiClock, bitOrder, dataMode)
In file included from sketch\src\HAL\HAL_STM32F1\HAL_spi_Stm32f1.cpp:40:0:
C:\Users\....\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.2\libraries\SPI\src/SPI.h:111:2: error:   initializing argument 2 of 'SPISettings::SPISettings(uint32_t, BitOrder, uint8_t)' [-fpermissive]
  SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {
exit status 1`

The code of the original HAL from victor_pv don't have this fonction : 'void spiBeginTransaction(uint32_t, uint8_t, uint8_t)'
https://github.com/victorpv/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_STM32F1/HAL_spi_Stm32f1.cpp

I think i'm close to compile it and it would be great to compile it with arduino because it's really simple with the STM32duino bootloader to upload sketchs, it's really like uploading a 2560 sketch.  If somebody could help me it would be really cool.
Thanks

youprint, 2018-02-07:
I compiled it, and it's working with the old Victorpv SPI code. So the modification by revilor on 3rd January of this year is not good (he wrote "untested on STM32F1")
https://github.com/MarlinFirmware/Marlin/pull/9019

thinkyhead, 2018-02-17:
@revilor — Is this resolved now?

alexxy, 2018-02-18:
No. Threre are missing spiBeginTransaction

dasaki, 2018-03-01:
in HAL_SPI.h:

#include <wirish_constants.h>

thinkyhead, 2018-03-04:
@dasaki — Do we need to enforce those platforms? And, do they provide environment vars that we can use for that purpose?

dasaki, 2018-03-04:
I was testing if it would be possible to make marlin work with the cheap "bluepill" (STM32F103C8T6) and make a motherboard, but it seems the code won't fit (code size is 130% of the board's 64k, I compiled in the arduino environment after installing the "Arduino SAM boards" via the board manager ).
The Maylan 200 is using a STM32F103CBT6 with 128k, so I may get one of those to try.
Thank you for your great work.

boelle, 2019-02-20:
@revilor is this one resolved/fixed?

@youprint tried latest bugfix 2.0 ?

closed 2019-07-06 by shitcreek