MarlinFirmware/Marlin #9477: MKS sbase digipots not taking any notice of the current values

adamd76, 2018-02-04:
------------------------------
### Bug Report
- Description: MKS sbase digipots not working
- Actual behaviour: 
When changing current values in configuration_adv.c this doesn't seem to have any effect on the drivers.
My extruder motor was getting very hot even though the current was set to 0.6A (same value I used in smoothie). I tried to lower the current but this wasn't making any difference.
I changed all current values to 0.1A to see if setting current was working at all:
#define DIGIPOT_I2C_MOTOR_CURRENTS { 0.1, 0.1, 0.1, 0.1, 0.1 } 
After that all motors were still pulling strong as though the drivers were still set to high current.
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1693087/configuration.zip)

curly-paul, 2018-02-04:
Also having the same issue, unless I should be changing values somewhere else for Sbase ?
[image]

Spawn32, 2018-02-04:
This is the correct place to change the values, i know this used to work, haven't re compiled any new source for a while, but have moved to a Tronxy X5S corxy and will need to use the sbase on it (that little tinny thing the call a controller on the X5S is just funny, LOL) , so should be able to test this in a day or two...

adamd76, 2018-02-04:
Spawn32 - I've just compiled from your fork with my configuration files and the digipots are working just fine in your version. It must be something broken in the new sources....
Thanks,
Adam

Spawn32, 2018-02-04:
Ahh, ok :)
That fork is from the 25 Dec, a lot has been added and fixed since then, one of the dev's will probably look at this soon ...

Can't remember that anything was done / modified to the digipots...

adamd76, 2018-02-05:
OK. I managed to track down the commit that brakes the digipots.
It's the commit from 24th Jan. Looks like there are some changes to mcp4451 digipot routine.
[image]

https://github.com/MarlinFirmware/Marlin/commit/45cfb1642493017b1203e35753fc258579b4d055

Prior to this commit digipots seem to work fine.

Spawn32, 2018-02-05:
Nice, should be easy for @thinkyhead to fix it then :)

adamd76, 2018-02-07:
Would be nice if he could fix it. For the time being I just removed the change from that commit and I've got the latest build working just fine.

Spawn32, 2018-02-07:
@tcm0116 
Anything you could have a look at ?
Probably affects all digipot boards, so it's more or less a major bug:/

tcm0116, 2018-02-07:
I don't have a board with digipots, so I can't really look at it.

Spawn32, 2018-02-07:
ah,, ok, to bad :)

Spawn32, 2018-02-07:
just got my X5S corexy running with the Sbase board, can confirm that they get very hot, almos like they are running on full amp's...

Spawn32, 2018-02-07:
@thinkyhead you should upgrade me so i can slapp labels on stuff, this one need a big red "BUG" on it, LOL

thinkyhead, 2018-02-17:
@adamd76 — Which part of #8355 seems to be causing the breakage? I'm just casually scanning issues to close and resolve the easier ones at the moment, but would like to come back to this later today.

adamd76, 2018-02-19:
@thinkyhead I don't know exactly which part of the code breaks it. All I know that when I removed the changes from that commit by replacing the files from the previous commit it all started working again.

Spawn32, 2018-02-20:
Is this being looked at ?
I know that @tcm0116 doesn't have a digipot device, so it's not easy for him to test :/
For us with digipot devices it's pretty big, it's  like having a older stepper driver with a missing pot 😄 

Could it also mess up the board during a print so the watchdog is triggered ?
I cant Finnish any prints, not from USB oand not from SD, totally random disconnects and board gets unresponsive..

https://github.com/MarlinFirmware/Marlin/issues/9593

And if i compile without watchdog i get compile errors...

thinkyhead, 2018-02-28:
Try this change in `Marlin/src/HAL/HAL_LPC1768/Wire.cpp`:
```diff
-  transferMCfg.sl_addr7bit = txAddress >> 1; // not sure about the right shift
+  transferMCfg.sl_addr7bit = txAddress; // Right-justified 7-bit address
```

As3uS, 2018-03-01:
Hi guys , i will try if i see amelioration, just with the motors noise i will know ^^ i give you a feedback if this fix solved the issue.
And thank all of you for your big work

thinkyhead, 2018-03-04:
Ah, well here's the problem. The old `Wire.h` code was being conceited. It was all about digipots and specific to MKS SBASE. The following lines were removed from the class, which also removed all initialization of the digipots.
```diff
-void TwoWire::beginTransmission(uint8_t sla) { digipot_mcp4451_start(sla);}
-void TwoWire::begin(void) {digipot_mcp4451_init();}
-size_t TwoWire::write(uint8_t data) {return digipot_mcp4451_send_byte(data);}
-uint8_t TwoWire::endTransmission(void) {return 1;}
```

thinkyhead, 2018-03-04:
@tcm0116 — Not sure who to ask, and I'll do more online research in the meantime, but…

Does it seem to you that the i2c code used by these digipots is specific only to the digipots and the bus(es) on the MKS SBASE board, or does it seem like it might generally apply to LPC and some other boards? I ask, because maybe for MKS SBASE the generic Wire class might not be needed — but then, it might be if a separate i2c bus is also desired.

closed 2018-03-04 by thinkyhead

thinkyhead, 2018-03-05:
Darn. Well, I'll soon recruit some MKS SBASE testers and do a debug session in my copious spare time.

As3uS, 2018-03-05:
i have compiled yesterday but i don't have the time to test, i will try tonight.

adamd76, 2018-03-05:
I will try to test it tonight.

curly-paul, 2018-03-05:
Can confirm latest build works regarding digipot current values.
Tested will all at 0.1 and now running with 'normal' values 
Thankyou

As3uS, 2018-03-06:
Hi can't try, it seems have heating problem on the last build, printer can't stabilize the temperature brefore start print :'(

thinkyhead, 2018-03-07:
> Can confirm latest build (from ~9:00am GMT) works regarding digipot current values.

I just knew it would work. Kudos to me! Meanwhile, still wondering why it doesn't work with the regular ol' I2C library. What is this strange voodoo?

commit beeed580 (2018-03-04): Implement digipots for MKS SBASE (#9927)

Fix #9477