MarlinFirmware/Marlin #9478: Rumba with MKS MINI12864 v1.0

billkaroly, 2018-02-04:
I am attempting to set up a Rumba board with a MKS MINI12864 v1.0 and I get the following error. I am way out of my league right now and do not know the difference between 8 Stripes or 4 either. I tried a search. Any help would be appreciated.

### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\ultralcd.cpp:96:0:

ultralcd_impl_DOGM.h:208: error: 'DOGLCD_CS' was not declared in this scope

   U8GLIB_MINI12864 u8g(DOGLCD_CS, DOGLCD_A0);  // 8 stripes

                        ^

ultralcd_impl_DOGM.h:208: error: 'DOGLCD_A0' was not declared in this scope

   U8GLIB_MINI12864 u8g(DOGLCD_CS, DOGLCD_A0);  // 8 stripes

                                   ^

exit status 1
'DOGLCD_CS' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

GMagician, 2018-02-04:
Your configuration please.

billkaroly, 2018-02-04:
Please pardon my ignorance. By configuration you mean what exactly. Basically I'm building similar to a Prusa. Using the Rumba controller and the MKS MINI12864 LCD dispay. I'm using the Geetech I3 Pro X  configuration.h file as a starting point. My bed is 214x214.

billkaroly, 2018-02-04:
I found this

#elif ENABLED(MINIPANEL)
   #define MISO_PIN      50  // system defined - only needed if using onboard SD card
   #define MOSI_PIN      51  // system defined
   #define SCK_PIN       52  // system defined
   #define SDSS          53  // only needed if using onboard SD card

   #define BEEPER_PIN    37

   #define BTN_EN1       31
   #define BTN_EN2       33
   #define BTN_ENC       35

   #define KILL_PIN      41  // optional

   #define SD_DETECT_PIN 49  // only needed if using onboard SD card

   #define DOGLCD_CS     25
   #define DOGLCD_A0     27

GMagician, 2018-02-04:
by configuration I mean configuration.h and configuration_adv.h files. Pack to zip and attach

Where did you find these notes? I don't know anything about your system so to let me do something I need cnf files

billkaroly, 2018-02-04:
OK. I'll attach shortly. Got the notes from [here](http://reprap.org/wiki/MKS_MINI_12864)

billkaroly, 2018-02-04:
I added the defines to pins_rumba.h and I am recompiling. So far no errors but I will attach zipped files anyway. thanks for the help.

billkaroly, 2018-02-04:
[rumba_MINI.zip](https://github.com/MarlinFirmware/Marlin/files/1693239/rumba_MINI.zip)

billkaroly, 2018-02-04:
It compiled but the display does not work. It lights up but thats it. Ive tried swapping cables around too. My understanding is EXP1 and 2 are labeled pin 1 on the right when it is actually on the left.

GMagician, 2018-02-04:
Sorry but I don't know how your H/W is so can't help you more

BlocksTechnology, 2018-02-07:
I had a similar problem. I will attach the issue I created and that was solved by @Bob-the-Kuhn

BlocksTechnology, 2018-02-07:
https://github.com/MarlinFirmware/Marlin/issues/8504

BlocksTechnology, 2018-02-07:
You just need to change the pins of DOGLCD_A0 and DOGLCD_CS

thinkyhead, 2018-02-17:
@BlocksTechnology — Anything more we need to do in the firmware to resolve this? Or has #8504 got it covered?

BlocksTechnology, 2018-04-03:
Sorry @thinkyhead did not see your question. #8504 got it covered

closed 2018-04-10 by thinkyhead

co8fs, 2019-03-10:
Hi Guys
I have the similar problem and I can't find DOGLCD_CS and DOGLCD_A0.
Any help please.

co8fs, 2019-03-10:
Or  Please, Can somebody shut me His conf file to co8fs@yahoo.com.
Thanks.