MarlinFirmware/Marlin #9480: [FR] Multiple Z-Probes

compatibilizer, 2018-02-04:
I believe it is a good time to discuss the implementation, since we now control separate hotends independently, have the ability to probe with the nozzle, and work on a unified bed leveling system. 

Obviously Marlin can only handle a single Z-probe at the moment, but probably there are a handful of applications where multiple probes could be useful. Below are the ones I came up with, feel free to add more applications here.

## Example 1: Automated Z-Offset calibration

This was the first time I realized multiple probes would be a good idea. I am working on independent dual-extruders, and every time I make a hotend reassembly, nozzle X-Y offset stays pretty much consistent, but Z-offset always requires calibration. 

I am using a [piezo z-probe](http://forums.reprap.org/read.php?1,767998) [integrated on top of the hotend.](http://forums.reprap.org/read.php?1,635075,page=16) It allows probing with a feather-light touch of the nozzle tip. It can be used on both hotends, then Marlin can probe a single point (e.g. center of bed) with each nozzle respectively and find out the Z-offset automatically. This example is probably the easiest to implement.

## Example 2: Proper bed leveling in Parking Extruders

This one is a little elusive at first, but bear with my simple sketches, they will help understand the issue. In a [parking extruder setup](http://diy3dprinting.blogspot.com.tr/2014/12/dual-parking-extruder-by-3d-proto.html), carriage position with respect to nozzle changes for each tool.

[image]

Now we usually think about bed leveling as if we're compensating for the bowing of the build plate only, It is actually the bowing of the build plate and bowing of the linear rods combined. This carriage offset in parking extruder setup causes a different ABL grid for each tool. It is not the same grid with some offset in X direction, it is a completely different grid, because you're combining a different pair of points on linear rod and build plate.

[image]

You can also encounter this error in a single nozzle setup if you mount your probe too far away from your nozzle, but that doesn't happen too often, we're usually placing our probes as close to our nozzles as possible. That is why piezo probes are probably the best way to move forward, since they virtually have no offset between the probe and nozzle (I'm not even mentioning that they work on every build surface). 

Anyway, when you probe with one nozzle in a parking extruder setup, probing nozzle works perfectly, but second nozzle is usually too far away and ABL grid error increases so much that your first layer won't stick the build plate at certain regions. Right approach would be to probe with both nozzles at the start of a print, and use different ABL grids for each tool. 

## Example 3: Delta effector tilt compensation

I ask for your further elaboration on this one, because I am not using a delta, but I know how they work and that one of their biggest challenges is to align the effector perfectly parallel to the build plate. Multiple probes can allow Marlin to solve this issue automatically.

It should be possible to compensate for the effector tilt with some simple trigonometry and 3 probes at the corners of the effector . Just like auto bed leveling, we could call it "auto effector tilt leveling". We could even compensate for different length rods. That would be really interesting to watch :)

fiveangle, 2018-02-04:
Another mechanism — one which is a feature that I was considering looking into — is...

In standard single-probe cartesian, there is an offset from the nozzle which reduces the probing area to less than the full bed if the gantry movement limits match the printable bed limits (Printrbot Simple Metal for example).

With dual Z probes located on opposite sides of the nozzle, one could perform standard bed levelling calibration with the primary Z probe (with Z-offset, probe offset, Babystepping, Coordinate System Offset, or whatever total Z-axis coordinate stack combo is relevant for the specific config being used) and then use it to calibrate the secondary probe by probing the same point with each probe and recording the delta.

Not only would you then hit all extremes of the bed, but also probe the bed in a bit over 1/2 the time, since you would cut the XY movements during probing to 1/2.  Worst case scenario is you have to probe 2 full times at each XY coordinate, although it may be possible to monitor both Z probes simultaneously and capture Z for each on a single Z probe movement, making probe time exactly 1/2.

-=dave

compatibilizer, 2018-02-05:
I think both of your ideas are really good, especially the "extended probe reach" application, because it plagues many users. 

I remember when I used "interpolation" for those regions that are out of reach, it resulted in the nozzle scratching the build plate, and when the interpolation was turned off, layers did not stick there. I had to modify the gantry to extend the probe reach, and it still did not cover the whole bed. Then I switched to piezo probe and never looked back.

I have added your ideas to the original post as Examples 4 and 5, let me know if I have missed anything.

gcurtis79, 2021-05-22:
I have a printer with Dual Z motors, and an endstop on each wired in parallel as the Z_MIN_PIN endstop. It would be nice to separate them, and have marlin detect either Z_MIN or Z2_MIN (on Z_MAX_PIN) as probes.

Perhaps something like, instead of "#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" it could check both pins Z and Z2 when they are both defined as Z_MIN/Z2_MIN pins.

This would allow for auto-leveling of the Z towers so X movements are level, and allow probing to be triggered by either endstop depending on which side of the X-axis the nozzle is on.

fiveangle, 2021-05-22:
@boelle, I think the `F:Z-Probes` label is appropriate.  I suspect you accidentally removed it?

[image]

fiveangle, 2021-05-22:
> I have a printer with Dual Z motors, and an endstop on each wired in parallel as the Z_MIN_PIN endstop. It would be nice to separate them, and have marlin detect either Z_MIN or Z2_MIN (on [...]

But the bed is "levelled" referencing the towers.  In your scenario, there's no reference ?  The bed is known unlevel across the surface and with dual z-motors moving independently, there's no way to reference even the bed since movement since there are 2 degrees of freedom for each (with each Z-tower).  I'm sorry but I don't see any way to further constrain it so it can work you want.

What would be possible however is having a fixed probe point closest to each tower to set the towers to a fixed "homing point" (similar to how Prusa uses the digipot capabilities of their TMC drivers to run the towers up to the top physical endstops in extremely low power in order to "home" them to a fixed point, then they reference out from there) but it would require multiple iterations to eliminate the remaining degree of freedom introduced by the opposite tower.

I think your idea opens up the possibility for a lot of microvariables and in the end, really only benefits saving a pin on the board.  Certainly possible, but probably not the value-ad you're hoping for.

gcurtis79, 2021-05-22:
I might be totally misunderstanding, but I thought you could make the machine home a dual motor axis to the first end stop triggered, then continue moving only the un-triggered motor until it hits it's trigger point, thus 'leveling' that span.

When using this feature on the Z axis, you can't also use the 'Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN' feature, because Z_MIN and Z2_MIN both exist, and won't necessarily both be triggered when probing different areas of the bed.

The functionality I'm looking for is that in this case, where Z_MIN and Z2_MIN both exist, either one of them getting triggered acts as a Z_MIN_PROBE trigger.

fiveangle, 2021-05-25:
yes, but you are not using hard endstops, you are trying to connect these inputs to probes that are a variable distance from each lead screw, so if tower 1 triggers first, tower 2 is still at an unknown state.  if it is higher than tower 1, once you move the carriage to that side and home, it will be correct, but now tower 1 homing will be off due to the offset of the prob from the lead screw, which is the fulcrum point of the carriage when right tower homing operation occurs.  like I said, the only way to resolve would be to iterate back and forth until the difference is below desired error.  alternately, one could introduce yet another config option to specify PROBE_LEFT_LEFT_Z_SCREW_OFFSET and PROBE_RIGHT_Z_SCREW_OFFSET then use simple trig to calculate what the adjustments should be, but again, that would benefit such a few number of people and likely just confuse more people and become a support nightmare to explain + yet more config bloat that is almost never used.  If someone wants to build it, I'm sure it would be considered if it worked well, but I wouldn't hold my breath waiting for anyone to work on this ;)