MarlinFirmware/Marlin #9482: Bug : G2 Broken ?

dark-nico, 2018-02-04:
### Bug Report
- Description: 
Hi, 
I'm using the latest bugfix-1.1.x to control my CNC.
I tried to use gcode with arc G2/G3 but G2 does nothing. 
G3 behave like it should (CCW) but G2 doesnt move at all. 
I've got ARC_SUPPORT enabled. 
I tried ARC_P_CIRCLES and CNC_WORKSPACE_PLANES with no luck.

- Expected behaviour:
Moving G2
- Actual behaviour:
No movement with G2 command
- Steps to reproduce:
  - Home 
  - move axis anywhere with enough room.
  - G2 I20 J20
  - G3 I20 J20

Let me know if you need logs or anything.
BR, 

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/1693337/Marlin_config.zip)

thinkyhead, 2018-02-06:
Do you see the same issue with 1.1.8 or 1.1.7 at all?

dark-nico, 2018-02-08:
Hi, 
It's working fine (both G2 and G3) in 1.1.8 with the same config files (except NO_VOLUMETRIC, I get compile error see #9083 ).
I made another test in bugfix-1.1.x without NO_VOLUMETRIC to make sure it's not linked but G2 is still not working.
I think we can confirm the bug. 
Let me know what I can do to help.
BR,

thinkyhead, 2018-02-11:
This change to the `plan_arc` function should fix the `G2` issue:
```diff
- mm_of_travel = linear_travel ? HYPOT(flat_mm, linear_travel) : flat_mm;
+ mm_of_travel = linear_travel ? HYPOT(flat_mm, linear_travel) : FABS(flat_mm);
```

commit 61f30519 (2018-02-11): Fix plan_arc clockwise (G2)

Fix #9482
commit 37710861 (2018-02-11): Fix plan_arc clockwise (G2)

Fix #9482
closed 2018-02-11 by thinkyhead