MarlinFirmware/Marlin #9483: [BUG] 2.0.x bugfix, manual probe, slow XY feedrates

paukstelis, 2018-02-04:
Manual probing with bilinear mesh bed leveling has ridiculously slow XY movement speed when moving to the next grid point (not using LCD).

It is anticipated that XY feedrates should be as set with HOMING_FEEDRATE_XY.

To reproduce:
manual probe, bilinear mesh enabled.
Home.
G29 S1.
Wait a very long time for X and Y to move 20 mm.
G29 S2.
Wait, etc.

paukstelis, 2018-02-07:
Closing as this is corrected by using just MESH instead of BILINEAR, which is how it is intended to work. Perhaps a sanity check so PROBE_MANUALLY works only with UBL or MESH leveling?

closed 2018-02-07 by paukstelis

thinkyhead, 2018-02-14:
PROBE_MANUALLY was made specifically for `AUTO_BED_LEVELING_3POINT`, `AUTO_BED_LEVELING_LINEAR`, and `AUTO_BED_LEVELING_BILINEAR`. With `MESH_BED_LEVELING` and `AUTO_BED_LEVELING_UBL` it is ignored as redundant.