MarlinFirmware/Marlin #9487: Layer shifting

Tamonir, 2018-02-05:
Hi everybody,

- Description: 
i recently switched my MKS-Base 1.4 (TEVO Tarantula) for a MKS GEN L 1.0 with FYSETC TMC2130 on all axes/E0/E1. With MKS Base, everything worked fine, even on Long Printjobs - perfectly aligned, no steplosses - of course, I did't change any mechanics when changing the board.
Yesterday, I did the final calibration (steps/mm) and printed the first 20*20*20 test cube. As you can see in the Picture attached, I got "some" issues on my y-Axis while the X-Axis is perfectly fine (20.01 mm). Z-Height of cube is 18.5 mm I will look for that later).

[image]

Do you have any idea on what may cause the y-issues? In Detail, you can say that the whole layers are shifted so the shift is caused at the beginning of a layer. Within one layer, y-axis perfectly fits to the (shifted) Position. Shouldn't TMC2130 recognise the missmovement and compensate?

- Actual behaviour:
Whole Layers get shifted at the beginning of a layer

- Desired behaviour:
No shift :)

- Steps to reproduce:
  - MKS Gen L 1.0
  - all axes TMC 2130 (so far properly working)
  - Marlin Tarantula (1.1.8 with quick setup from here ( https://github.com/JimBrown/MarlinTarantula )
  - My current config:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1692639/Configuration.zip)
  - Test Cube sliced by simplify 3d with standard-Speed Settings (60 mm/s printing and 130 mm/s travel Speed ? I am not on the machine right now)

Thanks in advance

Tamonir

Athemis, 2018-02-05:
I saw something similar on my delta when I switched to TMC2130. My guess was that the driver current was too low, hence leading to skipped steps. After I increased the current, the problem went away.

Tamonir, 2018-02-05:
Hi Athemis,

thanks for your idea, I will test it today But as i dont have any step losses during the print of a layer, it cannot be a general step loss. I just talked to a colleague of mine who had the idea, that the accelleration (and thus deceleration) might be too strong to handle. This might also be related to the low current but drivers are allready getting very hot with 0.8 A. 

As this happens only when starting a new layer, the shift might be related to the high y-travel speed and hence strong accelleration. As accelleration (a) is proportional to Force (f) and force F is related to Current, I will try to decrease accelleration (which will in this case mean the same as setting a higher current). 

I will report back soon and I am open for further ideas :)

Regards

Tamonir

ikarisan, 2018-02-05:
@Tamonir I have similar problems. For me everything is working fine with 750mA... as long as the printer uses StealthShop. But when using StealthShop together with hybrid threshold (100mm/s) the layers are shifting on larger object. If I use spreadCycle instead of StealthShop the layers are not shifting.

Tamonir, 2018-02-06:
Hi everybody,

@ikarisan: Nice thought, I will try to disable hybrid mode and see what happens. Allready thought about that but forgot :)

For now, i lowered "a max X" and "a max Y" from 3000 (Standard) to 500 and that allready gave me huge improvements as you can see in the picture. Have some widening at the edges though but this might be related to the begins/ends of Extrusion which are distributed over all edges. With that low accelleration, the drives wont reach 100 mm/s so they arent changing to spreadcycle. 

[image]

I will do some further tests with old accelleration and forced stealth chop

@ikarisan: how fast can you go at 750 mA and what are your accelleration and jerk settings?

Best regards

Tamonir

ikarisan, 2018-02-06:
My values are set to:

#define DEFAULT_MAX_ACCELERATION      { 1000, 1000, 100, 10000 }
#define DEFAULT_ACCELERATION          400     // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration for travel (non printing) moves
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.3

But I can see that you have the same "bleeding corner" problem as I have... :( 
https://github.com/MarlinFirmware/Marlin/issues/9494

floyd871, 2018-02-11:
I can confirm this Issue, but only the Y-Axis is affected. Increasing the current to 1000 does not help. Decreasing the acceleration themes to help.

Roxy-3D, 2018-02-11:
It would be helpful if you checked the .gcode file to make sure everything is reasonable and there is no shifting.    Can you feed the .gcode file into:  http://gcode.ws/# and check that it's layers don't shift?

floyd871, 2018-02-11:
Just check it the gcode with http://gcode.ws/#, all fine

Roxy-3D, 2018-02-11:
OK...   That helps isolate the problem to the machine.   Does it shift the same way if you print multiple times?   I'm wondering if the stepper motors are losing steps????

floyd871, 2018-02-11:
Yes, the shft is always the same way Y-Axis increases, never decreases. 
When i increase the current the shifting is less, but overheating warnings in the console. Decreasing the acceleration to 1000 (from 3000) help. Shifting ist gone.

Tamonir, 2018-02-11:
@ikarisan: Thanks, I will try these higher amounts soon, I am now going with 500 accelleration and its perfect (despite the bleeding corners). I had some issues though (horizontal artifacts) which were caused by some layers beeing thicker than the others. This was caused by inconsistant z move (I am using 2 z steppers). I set the current for z to 1200 (isnt moving that much...) and now it works like a charm - despite bleeding corners.

Roxy-3D, 2018-02-11:
You also might want to try reducing your Max Feed Rates to 1/2 of their current values and see if that makes things better.   If it totally fixes the problem, you can start bumping the Feed Rates up until you start seeing the problem again.

floyd871, 2018-02-11:
And the Acceleration back to default 3000!?

Roxy-3D, 2018-02-11:
I wouldn't do that...    Even with a very low acceleration, the extruder gets up to speed pretty quickly.

floyd871, 2018-02-11:
I think this is really a problem with the TMC2130. Left one 800mA (skipping Z-Axis) , Right 1000mA (No skipping)
[image]

Tamonir, 2018-02-11:
> I thought Z axis is not so so problem with current, because of the reducers of the leadscrew..

And maybe thats the reason Maybe the very short turning of the steppers leads to stalling. Yesterday, I changed the microstep settings of y and x from 16 to 32 - so the distance got devided by 2. What was remarkable was, that the sensorless homing triggered immediately when homing (which was calibrated well for 16 steps) - as sensorless homing is provided by motor current, maybe that the load vor minor changes causes the stepper to stall. Its only a guess...

@ floys: do you have two or one z stepper?

floyd871, 2018-02-12:
Good point, yes i am use standard Ramps with two motors on one driver. 
So the default setting to 800mA for z-axis might not be enough....

Lyr3x, 2018-03-29:
I'm running my Z axis @700mA  without any shifting problems (nearly the same acceleration, jerk and feedrate). Do you use stealthchop or spreadcycle?

You need fairly high currents with the TMC2130 in stealthchop mode to prevent skipping steps

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-07 by Tamonir