MarlinFirmware/Marlin #9490: Print freeze midpring

rjcuomo1, 2018-02-05:
Hello,
When I print anything over 41mm tall, it freezes mid print at about 41mm. It's like I pressed the pause button. I can still scroll through the menus just fine. It doesn't matter how large or small the print is on the x and y axis. The heaters remain on and all the fans are still operational. It's almost like the Gcode to the print to stop and release the steppers; doesn't part the nozzle and it doesn't lower the bed.. Another thing that happens when it stops is that the LEDs turn off. They're controlled by the RUMBA board/Marlin. I've tried printing off octoprint and off straight SD and either way, if it hits about 41mm, it stops. I have double checked the build place sizes in both Marlin and Cura and they're both set to 356mm which is correct for the size of this printer.  Does anyone have any insight on why this might be happening?

Slicer: Cura 3.1
Board: BIQU RUMBA
Firmware: Marlin 1.1.7 bugfix
Printer: Hypercube Evolution (Core XY)

[config and config adv.zip](https://github.com/MarlinFirmware/Marlin/files/1696389/config.and.config.adv.zip)

thinkyhead, 2018-02-05:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch in case the issue is already fixed. If we already got it then you can close the issue. If you still see the bad behavior then we can investigate further.

rjcuomo1, 2018-02-06:
I put 1.1.8 bugfix on and it finishes the first layer then it stops the same way like it was doing at 41mm. I made a video which I will upload to Youtube in a bit so you can see what happens

rjcuomo1, 2018-02-06:
Here's the video:

https://youtu.be/zANJw4s8XL0

It seems like it happens at the layer change

thinkyhead, 2018-02-17:
Two things you can try:
- Reducing your Z micro-stepping by 1/2 so that you can reduce your Z steps-per-mm from 1600 steps per mm down to 800.
- Reduce your Z max feedrate to 4mm/s or 3mm/s.

boelle, 2019-02-20:
@rjcuomo1 tried what thinky suggested?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

Mahdi-Massahi, 2020-05-30:
I have kinda same problem, I notice mine is coreXY as @rjcuomo1. The printer stops after some time without any error or warning. The thing I got is, maybe we are having some memory issues handling coreXY system. Does marlin supports any errors related to memory or some thing?
I haven't tested what @thinkyhead mentioned yet, ill comment the results soon.

Mahdi-Massahi, 2020-06-25:
I decreased micro stepper settings from 1/16 to 1/4 using stepper drives' jumpers for all 3 axis and extruder as well. (also did corresponding modifications on config. file) And I do not have any freezing issues. The printer works properly now.