MarlinFirmware/Marlin #9498: Mind-bent, stuck on "Z_SAFE_HOMING_Y_POINT can't be reached by the Z probe"

filipgoc, 2018-02-06:
### Compile Error
When I compile with `Z safe homing` I get an error:
```
Z_SAFE_HOMING_Y_POINT can't be reached by the Z probe
```

I know I must be missing something obvious, but I just can't see it. Please lend me your fresh eyes. 

---

I tried to go through the calculations, and it don't add up: 

I am running a cartesian printer with the z probe behind and on the right of the extruder. 

Variables: 
```
======== Probe
=
=
=
=
Extruder

X_PROBE_OFFSET_FROM_EXTRUDER 22
Y_PROBE_OFFSET_FROM_EXTRUDER 25

X_BED_SIZE 180
Y_BED_SIZE 170

X_MIN_POS -13
Y_MIN_POS 0

X_MAX_POS X_BED_SIZE + 5
Y_MAX_POS Y_BED_SIZE + 5

Z_SAFE_HOMING_Y_POINT (50) // this is the issue. made comically large to satisfy equations... 
```

Following the sanity check, we ned min_probe points. 

```
#define _MIN_PROBE_Y (max(Y_MIN_BED, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER)) = 0, -2 + 25 = 23
#define _MAX_PROBE_Y (min(Y_MAX_BED, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER)) = 160, 165 +25 = 190
```
(Since this is a cartesian printer, then X/Y centers are X/Y_CENTER ((X/Y_BED_SIZE) / 2), and it follows that X/Y_MIN_BED=0 and that X/Y_MAX_BED are the same as X/Y bed size. )

then we check the sanity check: 
```
#elif !WITHIN(Z_SAFE_HOMING_Y_POINT, MIN_PROBE_Y, MAX_PROBE_Y)
    #if HAS_BED_PROBE
      #error "Z_SAFE_HOMING_Y_POINT can't be reached by the Z probe."

#elif !WITHIN(50, 23, 190)
    #if HAS_BED_PROBE
      #error "Z_SAFE_HOMING_Y_POINT can't be reached by the Z probe."
```
 
Since 50 is within 23 and 190, what is the problem here?
Maybe I don't see how homing directions come into play with Min and Max?
(Yet, I tried both just to make sure). 

I'm leaving in my config file for the curious mind! Thanks in advance. 

---

Attach a ZIP of `Configuration.h`
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1697556/Configuration.h.zip)

tofifee, 2018-02-06:
Just out of curiosity:

Why is your X_Pos defined outside your bed?

thinkyhead, 2018-02-06:
No worries. It's fine to have your `*_MIN_POS`/`*_MAX_POS` outside the bed. These are the physical limits.

thinkyhead, 2018-02-06:
I get a different error compiling 1.1.8 with your `Configuration.h`. But then, I don't have your `Configuration_adv.h`, if that's important.

```
In file included from /var/folders/9l/_vm2dfnn1w3dgmjfjgnzc8tc0000gn/T/arduino_build_425590/sketch/MarlinConfig.h:40:0,
                 from /var/folders/9l/_vm2dfnn1w3dgmjfjgnzc8tc0000gn/T/arduino_build_425590/sketch/G26_Mesh_Validation_Tool.cpp:27:
SanityCheck.h:793: error: static assertion failed: UBL_PROBE_PT_2_Y can't be reached by the Z probe.
     static_assert(WITHIN(UBL_PROBE_PT_2_Y, MIN_PROBE_Y, MAX_PROBE_Y), "UBL_PROBE_PT_2_Y can't be reached by the Z probe.");
     ^
SanityCheck.h:794: error: static assertion failed: UBL_PROBE_PT_3_Y can't be reached by the Z probe.
     static_assert(WITHIN(UBL_PROBE_PT_3_Y, MIN_PROBE_Y, MAX_PROBE_Y), "UBL_PROBE_PT_3_Y can't be reached by the Z probe.");
     ^
```

thinkyhead, 2018-02-06:
After this change it compiles:
```diff
- #define UBL_PROBE_PT_2_Y 20
+ #define UBL_PROBE_PT_2_Y 25
  #define UBL_PROBE_PT_3_X 120
- #define UBL_PROBE_PT_3_Y 20
+ #define UBL_PROBE_PT_3_Y 25
```

filipgoc, 2018-02-06:
Ha. Blimey. 

I guess the real question is then what the heck I was compiling late last night in my delusional state. 
Hard to know. Probably combined some who knows dev branch with who knows dev config. 

Thanks. 

---

@tofifee Yes, the min_pos is outta bed because the endstop is actually outta bed. (Gives me space for nozzle wiping gizmos and such.)

filipgoc, 2018-02-06:
PS: If that wasn't self-evident, I don' get that error anymore either. Eh..

closed 2018-02-06 by filipgoc