MarlinFirmware/Marlin #9507: How to center UBL G29 leveling grid with an offset bed?

ianmackinnon, 2018-02-06:
I have a Cartesian printer whose bed is not aligned to X and Y end-stops, so I use `M206` to set a home offset and align the edges of the bed to zero in printing space.

Quoting @thinkyhead from issue #9418.

> M206 shouldn't [...] have any effect on internal operations

Does that mean that the Unified Bed Leveling functions controlled by `G29` ignore home offsets set by `M206`? Is that also true for individual probe issued by `G30`?

I have done some tests using `bugfix-1.1.x` and this does appear to be true for both.

If that is the case, how can one center a leveling mesh on a bed whose physical X and Y edges are not aligned with end-stops and requires a non-zero home offset?

Roxy-3D, 2018-02-06:
A decision was made a while ago to allow all of the bed leveling systems to use native, machine coordinates.   If you are going to shift the home offset, that is done at a very high level and the lower level software (such as bed leveling) won't be aware of it.   You will need to use machine coordinates when tuning those sub-systems.

You can do a M206 and then use UBL's G29 W to see the mesh point locations.   They should stay the same before and after the M206.

ianmackinnon, 2018-02-06:
That makes sense.

So in the following line from `Configuration.h`:

    #define MESH_INSET 5              // Mesh inset margin on print area

Am I right in thinking that “print area” here has the width and height defined by `X_BED_SIZE` and `Y_BED_SIZE`, but it's aligned to the end-stops, not necessarily the bed itself?

In that case, would I be able to solve the problem by hard coding my bed's offsets into `X_MIN_BED`, etc. in `Conditionals_post.h`? Would that be safe?

ianmackinnon, 2018-02-07:
I defined the XY offset using two new macros `BED_OFFSET_X` and `BED_OFFSET_Y` in `Configuraion.h` and altered the following lines in `Conditionals_post.h`.

    #define X_CENTER ((X_BED_SIZE) / 2) + BED_OFFSET_X
    #define Y_CENTER ((Y_BED_SIZE) / 2) + BED_OFFSET_Y

The mesh is now physically aligned with the bed and I have not observed any adverse effects.

closed 2018-02-07 by ianmackinnon

Roxy-3D, 2018-02-08:
> Am I right in thinking that “print area” here has the width and height defined by X_BED_SIZE and Y_BED_SIZE, but it's aligned to the end-stops, not necessarily the bed itself?

Yes.    And the mesh inset subtracts some amount of space off of the bed to insure the nozzle does not run into any clamps you use to hold things like glass to the bed.