MarlinFirmware/Marlin #9511: Nozzel offset calcluated incorrectly

dchosking, 2018-02-06:
### Bug Report

Nozzel Offset from probe trigure point is incorrectly calculated

- Description: When setting up an auto level probe, the offset is not calculated the same if you skip a G29, this causes a bedcrash if you include a G29 in the .gcode but prints at the perfect layer hight if you dont add the G29

- Expected behaviour: The offset should be basically the same regardless of method used

- Actual behaviour: The offset is off by alot, in my case after just g28 -5.xx and after g28 -0.xx

- Steps to reproduce:
1. M851 Z0 - Reset Z0Offset
2. M500 - Store setting to eeprom
3. M501 - Set active parameters
4. M503 - Display Active Parameters
5. G28
6.1 G29 (Skip this on one attempt and you wont get the same offset)
7. G1 F60 Z0 - Move nozzle to true 0 offset
8. M211 S0 - Switch off soft endstops
9. Move nozzle towards bed slowly until the paper can barely move
10. Take note of the Z on the printer display 

My Configuration
https://pastebin.com/9atKJxsN

thinkyhead, 2018-02-17:
If you skip `G29` then the printer is just using the Z=0 obtained from `G28` and no extra correction is done to the XYZ position.

`AUTO_BED_LEVELING_LINEAR` is only recommended for bed that are known to be perfectly flat.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

dchosking, 2018-02-17:
I tested it with the latest bugfix 1.1.x (downloaded 2018/02/07)

what it seems like is happening is when you calculate the offset by going through the process of lowering the nozzel until it is the correct hight off the bed, if you only do a G28 there is an offset of exactly 5mm + nozzel distance, but when you do the calibration with the G29 then this offset of 5mm is not taken into account. But what this does mean is if i try and print Gcode that contains no G29 command it will will either print in the air or crash into the bed depending on which offset you used.

thinkyhead, 2018-02-25:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

You can use `M114 D` as you move around the bed to see lots of details about the current position. If you see strange things, also copy them into your text file.

Since you're using `LINEAR` leveling, you can also use `G29 Q` at any time to see details about leveling at the current position.

We look forward to your next report. Hopefully we can narrow this down to a single cause and fix it.

boelle, 2019-02-20:
@dchosking new report?

closed 2019-07-06 by shitcreek