MarlinFirmware/Marlin #9513: [2.0.x BUG] Screen freeze when SDSUPPORT enabled

oddwick, 2018-02-06:
### Bug Report
- Description: 

when  `SDSUPPORT` is enabled on RAMPS14 with viki2, miniviki, and azsmz_12864, the screen will freeze at different places in the boot process.  

with sd card, stuck on splash
without sd card, goes to info screen and freezes on first flash of XYZ

in both cases, the board is active and can send commands through serial, though things like contrast echo ok, but no effect to the screen. 

with  `SDSUPPORT` disabled, everything works as expected.

This is an ongoing bug and this morning i went through and compiled and tested all releases since 1.0.0 to trace back where this bug first occurs using the same hardware and settings
- 1.0.0 - everything works, a few artifacts on screen, but otherwise acceptable.  screen active and able to read from sd card.
- 1.0.1 - same as 1.0.0, without artifacts on screen and {seemingly} better refresh rates
- 1.0.2 - as 1.0.1
- 1.0.3 - as 1.0.1
- 1.0.4 - as 1.0.1
- 1.0.5 - as 1.0.1
- 1.0.6 - fail to compile, didn't bother to try and trace error, just moved on
```
In file included from sketch/ultralcd.cpp:79:0:
		sketch/ultralcd_impl_DOGM.h: At global scope:
		ultralcd_impl_DOGM.h:188: error: 'DOGLCD_SCK' was not declared in this scope
  		U8GLIB_ST7565_64128n_2x_VIKI u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI
                                    		^
		ultralcd_impl_DOGM.h:188: error: 'DOGLCD_MOSI' was not declared in this scope
  		U8GLIB_ST7565_64128n_2x_VIKI u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI
                                                		^
		exit status 1
		'DIODOGLCD_SCK_RPORT' was not declared in this scope
```

- 1.0.7 - this is where lcd begins freezing with `SDSUPPORT`.  screen almost totally black and text only visible at extreme angles.  serial monitor reports sd init fail when no sd card is inserted and sd ok with a card.  even with the sddetect pin enabled, it tries to initialize sd card on boot twice
- 1.0.8 - as 1.0.7
- 2.0.x_bugfix - first try, failed to compile.
```
In file included from sketch/src/lcd/ultralcd.cpp:108:0:
		sketch/src/lcd/ultralcd_impl_DOGM.h:193:55: error: 'DOGLCD_CS' was not declared in this scope
     		U8GLIB_64128N_2X_HAL u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI
                                                       		^
		sketch/src/lcd/ultralcd_impl_DOGM.h:193:66: error: 'DOGLCD_A0' was not declared in this scope
     		U8GLIB_64128N_2X_HAL u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI
                                                                  		^
		exit status 1
		Error compiling for board Arduino/Genuino Mega or Mega 2560.

```

when and added the following to `pins_RAMPS.h` after enabled VIKI2 
```  
#elif ENABLED(AZSMZ_12864)
      #define BEEPER_PIN       35

      // Pins for DOGM SPI LCD Support
      #define DOGLCD_A0        47
      #define DOGLCD_CS        32
      #define LCD_SCREEN_ROT_180

      #define BTN_EN1          45
      #define BTN_EN2          41
      #define BTN_ENC          43

      #define SDSS             53
      #define SD_DETECT_PIN    49 //-1  // Pin 49 for display sd interface, 72 for easy adapter board

      #define KILL_PIN         33

      #define STAT_LED_RED_PIN 39
      #define STAT_LED_BLUE_PIN 37
```

as 1.0.7

this was all tested on ramps 1.4, with azsmz_12864 controller and default configuration files with the following changes

Configuration.h
```  
#define MOTHERBOARD BOARD_RAMPS_14_EFF
#define SDSUPPORT
#define VIKI2 (1.0.x)
#define AZSMZ_12864 (2.0.x)
```

pins_RAMPS.h
```
      #define BEEPER_PIN       35

      // Pins for DOGM SPI LCD Support
      #define DOGLCD_A0        47
      #define DOGLCD_CS        32
      #define LCD_SCREEN_ROT_180

      #define BTN_EN1          45
      #define BTN_EN2          41
      #define BTN_ENC          43

      #define SDSS             53
      #define SD_DETECT_PIN    49 //-1  // Pin 49 for display sd interface, 72 for easy adapter board

      #define KILL_PIN         33 (AZSMZ_12864)

      #define STAT_LED_RED_PIN 39 (VIKI2)
      #define STAT_LED_BLUE_PIN 37 (VIKI2)
```
so as far as i can tell, this has been an issue since 1.0.6~7.  has there been any major changes to the SPI / SD code during that time?

### New config settings:

Obviously this has **never** been tested with this combination of hardware because the pins given are an impossible combination for viki2, miniviki, or azsmz and seem to have been replicated each release.  see above for the correct pinout with the differences that the viki/miniviki has LED status and azsmz has a kill button.  

in addition in `Conditionals_LCD.h`, the proper settings for AZSMZ_12864 contrast should be:
``` 
#elif ENABLED(AZSMZ_12864)
    #define LCD_CONTRAST_MIN  20
    #define LCD_CONTRAST_MAX 80
    #define DEFAULT_LCD_CONTRAST 65
    #define U8GLIB_ST7565_64128N
  #endif
```
anything beyond 80 pretty much blacks out the screen and 65 is a good solid contrast (at least on ramps anyway)

but these setting work fine on this hardware all the way back to 1.0.0

Bob-the-Kuhn, 2018-02-06:
I was able to reproduce your problem with bugfix-2.0.x.

There's a similar problem with bugfix-1.1.x except that the SD card isn't functional.

oddwick, 2018-02-06:
Great!  thought i was losing my mind!  Did you also note the pin assignments to get it working?  in the pins config it lists 22 and 7 and on ramps, neither are accessible.   if you want, after work i can send the modified files your way.

oddwick, 2018-02-07:
Here are the modified files
```
Conditionals_LCD.h
pins_RAMPS.h
Configuration.h
```
[AZSMZ_ramps_mods.zip](https://github.com/MarlinFirmware/Marlin/files/1701789/AZSMZ_ramps_mods.zip)

Bob-the-Kuhn, 2018-02-08:
Sorry - I didn't have much time to work on this today.

I was able to get it to work by changing line 191 in ultralcd_impl_DOGM.h
DELETE:  #if DISABLED(SDSUPPORT) && (DOGLCD_SCK == SCK_PIN) && (DOGLCD_MOSI == MOSI_PIN)
REPLACE WITH: #if (DOGLCD_SCK == SCK_PIN) && (DOGLCD_MOSI == MOSI_PIN)

There's more needed for a permanent fix but that should get you going.

oddwick, 2018-02-08:
Thanks for the work Bob!
the good news is that it works, with a few side effects, don't know if same thing happens on your end?

when booting without sd card, screen gets partial refresh between info screen and splash where the top quarter goes to info screen while bottom is still stuck on splash and about a second or so then it refreshes.  
if a card is inserted, the screen says card inserted, but can't read from it in the menu.

if booting up with sd card, then boot is normal and card can be read from menu, but if card is removed, the contents of the card still show up in the menu and if selected, then things go haywire.

the most interesting thing is if it is only powered by 12v, then there is nothing on the screen and i can't get anything else to show up without having to refresh the whole unit.  don't know if this is a hardware issue, but same thing happens on 2 different boards with different screens, but on same boards with 1.1.5, this doesn't happen.  haven't tried it with anything later than 5, but going to try and see if your fix can be applied to 6+.

also, the blank screen issue pops up again if i try to use aux2 for any of the pins, any particular reason for that? 

speaking of pins, here is another pins_RAMPS.h with a more logical pin assignment for azsmz_12864.
[pins_RAMPS.h.txt](https://github.com/MarlinFirmware/Marlin/files/1708366/pins_RAMPS.h.txt)

thinkyhead, 2018-02-17:
>
```diff
-  #if DISABLED(SDSUPPORT) && (DOGLCD_SCK == SCK_PIN) && (DOGLCD_MOSI == MOSI_PIN)
+ #if (DOGLCD_SCK == SCK_PIN) && (DOGLCD_MOSI == MOSI_PIN)
```

@Bob-the-Kuhn — Do you recommend this as a permanent change?

Bob-the-Kuhn, 2018-02-17:
I'm not seeing any side effects when using VIKI2, mega2560, RAMPs & bugfix-2.0.x.

Bob-the-Kuhn, 2018-02-17:
I've just created PR #9686 to move the **DISABLED(SDSUPPORT)** change into 2.0.x

Bob-the-Kuhn, 2018-02-17:
@oddwick - there was a change in the VIKI2 driver between 1.1.5 and 1.1.6.  Let me know if the latest bugfix-1.1.x has the same side effects.  There may be a difference in the initialization portions of the drivers.

oddwick, 2018-02-18:
i just got through trying the bugfix-2.0.x and it works like a champ on ramps with azsmz with a few small changes. on first try, the contrast completely blacked out the screen, so in ```Conditionals_LCD.h``` i used:
```
  #elif ENABLED(AZSMZ_12864)
    #define LCD_CONTRAST_MIN  20
    #define LCD_CONTRAST_MAX 120
    #define DEFAULT_LCD_CONTRAST 60
    #define U8GLIB_ST7565_64128N
  #endif
```
this will take an azsmz from nearly blank screen to almost fully blacked out and 60 is about ideal.

and in ```pins_RAMPS.h``` on line 472  the following needs to be added in order for AZSMZ to work:
```    #elif ENABLED(AZSMZ_12864)

      // uncomment for buttons on left side
      //#define LCD_SCREEN_ROT_180

      // Pins for DOGM SPI LCD Support
      #define DOGLCD_CS        32
      #define DOGLCD_A0        47

      #define BTN_EN1          45
      #define BTN_ENC          43
      #define BTN_EN2          41
      #define BEEPER_PIN       39
      #define KILL_PIN         37
      
      // SD support pins
      #define SDSS             53
      #define SD_DETECT_PIN    49
```
now i am getting ready to try bugfix-1.1.x and see if that works

also @Bob-the-Kuhn, are you using pin config as is? because from what i can tell that is an impossible config for a viki2.
[Viki2_Wiring v1.2.pdf](https://github.com/MarlinFirmware/Marlin/files/1734917/Viki2_Wiring.v1.2.pdf)

oddwick, 2018-02-18:
just tried bugfix-1.1.x and it works out of the box with similar changes as above (using VIKI2 config with AZSMZ_12864 on ramps) with no observed issues.  

@Bob-the-Kuhn the ```ultralcd_st7565_u8glib_VIKI.h``` you posted does not work.  same results, the screen was shifted 4px right (buttons on my controller are on right side) and ignored screen rotation - screen appeared upside down - and would not refresh rightmost pixels.  the original works just fine.

oddwick, 2018-02-18:
update: i was just messing around with it some more and have noticed some errant behavior.  this only happens when ``SD_DETECT_PIN`` is enabled.  when set to -1, all is good.  when it is set to a pin, boots are super slow with a partial refresh, and the card is not detected when inserted or removed.  

it probably would be easier just to make a note that ``SD_DETECT_PIN`` should not be used rather than try and trace it for a relatively obscure hardware combination...

Bob-the-Kuhn, 2018-02-18:
Here's a different setup.  Another user reports this makes his display happy.

[ultralcd_st7565_u8glib_VIKI (0x14 col_adr).zip](https://github.com/MarlinFirmware/Marlin/files/1735091/ultralcd_st7565_u8glib_VIKI.0x14.col_adr.zip)

Bob-the-Kuhn, 2018-02-18:
Sorry, I was confused between two similar issues.

When using a AZSMZ_12864 
- Bugfix-2.0.x works but needs the LCD contrast change.
- Bugfix-1.1.x works but needs the LCD contrast change.
- In both cases there's problems when SD_DETECT_PIN is enabled.

Double check that you have the correct SD_DETECT_PIN selected.  

You might also try the following:
1. Enable PINS_DEBUG in Configuration_adv.h
2. Issue **M43 I** and see a) if it reports the correct pin and b) if there is a second logical signal attached to the pin.
3. Issue **M43 W** and short the SD_DETECT_PIN to ground.  The host display should report a pin number when you connect/disconnect the short.  It's safer if you use a 1K resistor rather than a direct short.

Bob-the-Kuhn, 2018-02-19:
@oddwick , @jeffyl - your AZSMZ_12864 contrast settings are very different.  

oddwick
```
    #define LCD_CONTRAST_MIN  20
    #define LCD_CONTRAST_MAX 120
    #define DEFAULT_LCD_CONTRAST 60
```

jeffyl
```
    #define LCD_CONTRAST_MIN  120
    #define LCD_CONTRAST_MAX 255
    #define DEFAULT_LCD_CONTRAST 190
```

I'm wondering if one of you is running this display at 3.3V and the other at 5V.  I can't find anything that tells me if this display is 5V only, 3.3V only or switchable.

jeffyl, 2018-02-19:
I tested with both an AZSMZ Mini and AZSMZ Smart RAMPS which are 3.3V boards. According to AZSMZ schematics, it is 5V power, but I assume 3.3V logic.

oddwick, 2018-02-19:
sorry about the confusion, but messed around with it this morning and here it is:

- Bugfix-2.0.x works but needs the proper pin assignments for ramps/azsmz12864
- Bugfix-1.1.x works but needs the proper pins for viki2
- both cases, sddetect does not work properly
- contrast can work with an expanded range maybe?

there is so little documentation for this screen it is hard to make heads or tails of it.  i am using it on ramps 1.4 and azteeg x5 mini, which ramps is a 5v board, and the azteeg outputs 5v, but I'm unsure if it uses 5 or 3.3 logic.  the only documentation i found for azsmz shows +5 for power input, and on the back there is no LLC, but there is a bank of resistors. if that was the case it would seem that 3.3 wouldn't be enough to give a logic high?

as for the pins, this is a working pin assignment from 1.1.5 where the sd detect pin works and has been tested fine.  in order to run it on ramps, you need to make custom cables, so they're laid out in three logical groups (buttons/lcd, spi/sd, serial)  so you don't have crossed wires between the two banks.

i tried to enable pins_debug and it won't compile in arduino 1.6.12 or 1.8.5.   i have to go out of town today so i won't get a chance to look at it until tonight.

Bob-the-Kuhn, 2018-02-19:
The 3.3V logic is not a problem.  Almost all 3.3V & 5V logic drive rail to rail so they drive the other just fine.

@jeffyl - please point me to the AZSMZ schematics.  I'd like to add them to my collection.

jeffyl, 2018-02-19:
I have two these images for the AZSMZ Mini: https://imgur.com/a/qhAU1
AZSMZ SMART RAMPS are on the reprap wiki: http://reprap.org/wiki/SMART_RAMPS

If you mean for the LCD, I do not have those. Is your collection uploaded anywhere?

thinkyhead, 2018-02-25:
Please let us know if pins debug still won't compile with the latest bugfix code.

oddwick, 2018-02-26:
Good news, pins debug compiles with no errors on ramps 1.4, and all works as expected.  with sddetect enabled, card is detected, read, and acknowledged when removed.  

whatever was changed with the lcd works just fine with the default contrast settings now with AZSMZ_12864.  was beating my head on desk trying to figure out why i was looking at a blank screen when i knew everything was connected properly... finally increased the contrast (i just when in before if compiled it and set it at 65 before i even began because before 190 would give a total black screen) and voila!  

which leads to the last point.  in the process of trying to figure out why i couldn't see anything on the screen i made a new cable and tweaked the pins for it so that they are sequential rather than logical.

so @thinkyhead if you could please add the following to ```pins_RAMPS.h``` on line ```455``` that should close this issue.  

there are no default pin definitions when using  ```#define AZSMZ_12864``` in ```Configuration.h``` for ramps so it doesn't work.  and the current pin definition for viki2 is just an impossible pin assignment.
on ramps 1.4, pins 22 and 7 are not even accessible on 1.4 and 7 is a servo pin on 1.3...
```
    #elif ENABLED(VIKI2) || ENABLED(miniVIKI)

      #define DOGLCD_CS         32
      #define DOGLCD_A0         47
      #define LCD_SCREEN_ROT_180

      #define BEEPER_PIN        35
      #define STAT_LED_RED_PIN  39
      #define STAT_LED_BLUE_PIN 37

      #define BTN_EN1           45
      #define BTN_EN2           41
      #define BTN_ENC           43

      #define SDSS              53
      #define SD_DETECT_PIN     -1 // Pin 49 for display sd interface
      //#define KILL_PIN          31

    #elif ENABLED(AZSMZ_12864)

      // uncomment for buttons on left side
      //#define LCD_SCREEN_ROT_180

      // Pins for DOGM SPI LCD Support
      #define DOGLCD_CS        41 
      #define DOGLCD_A0        45

      #define BTN_EN1          43
      #define BTN_ENC          37
      #define BTN_EN2          47
      #define BEEPER_PIN       35
      #define KILL_PIN         39
      
      // SD support pins
      #define SDSS             53
      #define SD_DETECT_PIN    49 //-1 // 49 enable for card detect
```

i also included my ```pins_RAMPS.h```

thinkyhead, 2018-02-26:
Duplicate of #8746

Let's move discussion to that issue.

thinkyhead, 2018-02-26:
Duplicate of #8746

closed 2018-02-26 by thinkyhead