MarlinFirmware/Marlin #9515: Probing with positive Z-offset issues?

paukstelis, 2018-02-07:
Not sure if this is a bug or simply an issue with my setup, so any insight would be helpful. 
I have a closed-loop DC motor printer and I am trying to use motor stalling of the nozzle going into the bed as a Z-probe. This requires a positive Z-probe offset since there is a small bit of slop in the coupler that has to be taken up before the stall is detected and the endstop is triggered. This value (0.5 mm) has been found empirically and it works consistently with a standard G28 Z. However, if I try to do any kind of auto leveling or even do an M48 the does multiple probings I get odd results. The initial probing is fine (0.02), but this follows:

`-1.051
-2.095
-3.171
-4.239
-5.291
-6.359
-7.403
-8.417
-9.469`

The position decreases the same amount (~1.04) with each probing. Eventually it fails when the current Z-value is below the -10 in the probing move. Its not clear to me where this is coming from but it seems as if the Z-position is not being properly registered after each probe. The Z-axis appears to be raising to the same height (3 mm) between each probing. Any insights appreciated.

AnHardt, 2018-02-07:
Your probing process is losing steps.
For homing this is ok, because the step-counters are reset there.
For probing this is a no go, because the lost steps add up.

paukstelis, 2018-02-07:
Hmm, I would have thought if I was losing steps that every time the probe raised it would get closer to the bed instead of going back up to 3 mm between probes. Thanks.

paukstelis, 2018-02-09:
Yeah, well this turned out to be one of those "user error" things. Old eyes no longer seeing decimal places in my dc motor servo controller. Closing and everything works as expected.

closed 2018-02-09 by paukstelis