MarlinFirmware/Marlin #9523: TMC2130 sensorless homing always in trigerred state

MarioStoilov, 2018-02-07:
- Description: M192 always reports endstops with TMC2130 to TRIGGERED even when motors are not running.

Using latest and greatest Marlin 1.1.8 release

```
18:42:14.768 : Reporting endstop status
18:42:14.768 : x_min: TRIGGERED
18:42:14.768 : y_min: open
18:42:14.768 : z_min: open

```
- Steps to reproduce:
  - Get a Watterott TMC2130 and hook it up to the X axis.
  - Send M192

Some additional info: I reworked the stallguard example from the TMC library in order to see if the drivers are the actual issue and added a read to see if the endstop is triggered. In the example, all is working as expected (when I stop the motor with my hand, the ENDSTOP pin goes to HIGH).
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1703970/Marlin.zip)
[StallGuard.zip](https://github.com/MarlinFirmware/Marlin/files/1703977/StallGuard.zip)

teemuatlut, 2018-02-07:
Have you tried `X_MIN_ENDSTOP_INVERTING`?

MarioStoilov, 2018-02-07:
Well, that worked.

Seems I need to read more on how the endstop logic works.

closed 2018-02-09 by MarioStoilov