MarlinFirmware/Marlin #9524: 'start_bmp' was not declared in this scope

JohnnyTheOne, 2018-02-07:
### Compile Error
When I compile with #9381 I get an 'start_bmp' error:
I use Anet_1.0 board it applies to the 1.1.x / 2.0.x

```
In file included from sketch\ultralcd_impl_DOGM.h:55:0,

                 from sketch\ultralcd.cpp:100:

dogm_bitmaps.h:649: error: 'start_bmp' was not declared in this scope

 static_assert(sizeof(start_bmp) == (START_BMP_BYTEWIDTH) * (START_BMPHEIGHT), "Bootscreen (start_bmp) dimensions don't match data.");

                      ^

dogm_bitmaps.h:603: error: 'START_BMPWIDTH' was not declared in this scope

   #define START_BMP_BYTEWIDTH ((START_BMPWIDTH + 7) / 8)

                                 ^

sketch\dogm_bitmaps.h:649:37: note: in expansion of macro 'START_BMP_BYTEWIDTH'

 static_assert(sizeof(start_bmp) == (START_BMP_BYTEWIDTH) * (START_BMPHEIGHT), "Bootscreen (start_bmp) dimensions don't match data.");

                                     ^

dogm_bitmaps.h:606: error: 'start_bmp' was not declared in this scope

   #define START_BMPHEIGHT (sizeof(start_bmp) / (START_BMP_BYTEWIDTH))

                                   ^

sketch\dogm_bitmaps.h:649:61: note: in expansion of macro 'START_BMPHEIGHT'

 static_assert(sizeof(start_bmp) == (START_BMP_BYTEWIDTH) * (START_BMPHEIGHT), "Bootscreen (start_bmp) dimensions don't match data.");

                                                             ^

dogm_bitmaps.h:603: error: 'START_BMPWIDTH' was not declared in this scope

   #define START_BMP_BYTEWIDTH ((START_BMPWIDTH + 7) / 8)

                                 ^

sketch\dogm_bitmaps.h:606:49: note: in expansion of macro 'START_BMP_BYTEWIDTH'

   #define START_BMPHEIGHT (sizeof(start_bmp) / (START_BMP_BYTEWIDTH))

                                                 ^

sketch\dogm_bitmaps.h:649:61: note: in expansion of macro 'START_BMPHEIGHT'

 static_assert(sizeof(start_bmp) == (START_BMP_BYTEWIDTH) * (START_BMPHEIGHT), "Bootscreen (start_bmp) dimensions don't match data.");

                                                             ^

exit status 1
'start_bmp' was not declared in this scope
```

JohnnyTheOne, 2018-02-08:
Enabled 
#define SHOW_BOOTSCREEN
#define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION 
and will works fine. but I do not use a bootscreen to save space for Anet board
same problem as teensy20 #9530

thinkyhead, 2018-02-17:
Patched!

commit 5c2ac018 (2018-02-17): Fix compile error with SHOW_BOOTSCREEN disabled

Fix #9530
Fix #9524
commit bf5df306 (2018-02-17): Fix compile error with SHOW_BOOTSCREEN disabled

Fix #9530
Fix #9524
commit 4ae9afcc (2018-02-17): Fix compile error with SHOW_BOOTSCREEN disabled

Fix #9530
Fix #9524
closed 2018-02-17 by thinkyhead