MarlinFirmware/Marlin #9525: Need Help Setting Probe Offsets for Bed Leveling

stu54, 2018-02-07:
I have a BL Touch on my Prusa. The probe tip is 35mm to the left of the nozzle and 15mm in front of the nozzle. My bed is 200mm by 200mm (and set in X_BED_SIZE and Y_BED_SIZE). Homing works fine and the nozzle is where I would expect it at (0, 0). The problem is with auto bed leveling. I am just going to talk about X maximum to keep it simple. In order for the probe to stay on the bed in the X axis, the maximum X would be 200 - 35 = 165 during bed leveling but should be able to print to 200mm because Marlin knew the probe was to the left of the nozzle. This meant that at X=200 during printing the probe would be off the bed but since autoleveling was already done it would not matter. What I expected is that I could compile Marlin using some combination of RIGHT_PROBE_BED_POSITION and X_PROBE_OFFSET_FROM_EXTRUDER that would autolevel probe at 200mm but know that the nozzle position is actually at 165mm. That would mean when Marlin built the bed level mesh it would remove the probe offset so the nozzle was considered to be at the probe location during printing.

Any help with this?

Roxy-3D, 2018-02-07:
Different bed leveling systems handle things differently.   You don't say which bed leveling system you are using.   If you use UBL, the unreachable (by the probe) positions are considered 'invalid' until you do something to give them a value.    Typically...   A G29 P3 is used to give them a 'reasonable' value and then a G26 mesh validation pattern is generated.    This pattern is used with a G29 P4 to edit those points to very accurate numbers.

stu54, 2018-02-08:
Thanks for your response. I am using bilinear because I wanted the process to be fully automated and just dependent on the values in Configuration.h

ManuelMcLure, 2018-02-09:
Are you sure the probe is to the left of the nozzle? From your description (when the nozzle is at X=200 the probe is off the bed) sounds to me as if the probe is to the right of the nozzle. Assuming the probe is actually to the right of the nozzle I get the following settings (assuming that the furthest left the nozzle can go to is 0 - if there's some extra space to the left like on my Anet A8 you can take advantage of that to probe further):

#define X_PROBE_OFFSET_FROM_EXTRUDER 35
#define Y_PROBE_OFFSET_FROM_EXTRUDER -15

#define LEFT_PROBE_BED_POSITION 35        // the same as X_PROBE_OFFSET_FROM_EXTRUDER
#define RIGHT_PROBE_BED_POSITION 195   // 5mm in from the edge to make sure we don't miss
#define FRONT_PROBE_BED_POSITION 5      // as above
#define BACK_PROBE_BED_POSITION 185    // bed size - Y_PROBE_OFFSET_FROM_EXTRUDER

#define EXTRAPOLATE_BEYOND_GRID         // extrapolate outside the grid

stu54, 2018-02-09:
Yes, the probe is to the left of the nozzle so when X=200 the probe is off the left side of the bed. The origin (home position) is the back right of the bed looking at the front of the machine. For bed level probing, X max would be 200 - 35 so the probe would be at 200 and the nozzle at 165. The probe being to the left of the nozzle also means that it can't probe at X=0 since the probe would be at X=35 when the nozzle is at X=0. My X limit switch would not allow the nozzle to move to X=-35 (off the bed to the right) to probe at X=0 so the leveling probe at X=0 is really X=35. I would not expect Marlin to be able to compensate Z during printing for X between 0 and 35.

ManuelMcLure, 2018-02-09:
OK - that explains my confusion. I'm more familiar with the origin being on the front/left of the bed. The nice thing is that because the origin is on the opposite corner, it means that the printer is just turned 180 degrees from what Marlin expects so you can simply imagine what the coordinates are if you look at it from the back. From that view you'd be seeing the origin be on the front/left as you're looking at it, which means that from that perspective the probe is to the right and behind the nozzle - I believe that this is what Marlin's world-view is as well. So given that, the X_PROBE_OFFSET_FROM_EXTRUDER, LEFT_PROBE_BED_POSITION, and RIGHT_PROBE_BED_POSITION settings above will work as desired, and we just need to adjust Y_PROBE_OFFSET_FROM_EXTRUDER, FRONT_PROBE_BED_POSITION, and BACK_PROBE_BED_POSITION:

#define X_PROBE_OFFSET_FROM_EXTRUDER 35
#define Y_PROBE_OFFSET_FROM_EXTRUDER 15

#define LEFT_PROBE_BED_POSITION 35 // the same as X_PROBE_OFFSET_FROM_EXTRUDER
#define RIGHT_PROBE_BED_POSITION 195 // 5mm in from the edge to make sure we don't miss
#define FRONT_PROBE_BED_POSITION 15 // he same as Y_PROBE_OFFSET_FROM_EXTRUDER
#define BACK_PROBE_BED_POSITION 195 // 5mm in from the edge to make sure we don't miss

#define EXTRAPOLATE_BEYOND_GRID // extrapolate outside the grid

The EXTRAPOLATE_BEYOND_GRID will allow bed leveling to make some smart guesses as to what the bed is like outside the probable area - if you leave it disables it will just consider the area outside the probable area to be at the same level as the closest edge of the grid.

thinkyhead, 2018-02-17:
> Yes, the probe is to the left of the nozzle so when X=200 the probe is off the left side of the bed. 

Ah, so yes, as @ManuelMcLure explains, "pretend" that the nozzle is at the "right" side of the bed for interpreting the instructions. In other words, rotate your printer 180° and configure it just as thought the origin 0,0 was "front-left" and it will all make sense.

closed 2018-02-17 by thinkyhead