MarlinFirmware/Marlin #9528: [FR] Mesh bed leveling + probe manually for Taz 4?

m-dan-7, 2018-02-07:
I'm looking through the Marlin_main.cpp files for the Taz 4 (Marlin 2016q2, version 1.0.0.1?) and comparing with that of the most recent Marlin version. The Taz 4 version allows for two types of automatic bed leveling, 3-point and "accurate bed leveling" which seems to be an older version of Mesh bed leveling. I'd like to be able to use this Accurate Bed Leveling configuration, but probe manually through g-code (or LCD) as is possible with the most recent Marlin. There isn't an option to 'probe manually' in the Taz 4 version though as far as I can see. Has anyone tried setting up manually probed mesh/'accurate' bed leveling for previous Marlin versions?

Firmware in question for Taz 4 located at - http://download.lulzbot.com/TAZ/4.0/software/2016Q2/firmware/Marlin/

thinkyhead, 2018-02-17:
The current Marlin includes TAZ 4 configuration files. If you upgrade to the latest Marlin using those and find issues then we can look into it. We do not support third-party forks.

m-dan-7, 2018-02-20:
@thinkyhead Where do I find the Taz4 configuration files? I was unsuccessful in trying to flash the latest Marlin to the printer, but I may be missing some steps in configuring it.

thinkyhead, 2018-02-25:
I'm sure you were able to find the TAZ configurations given 5 whole days. But just for the reference of those now visiting this page, they are in the folder:
```
example_configurations/AlephObjects/TAZ4
```
The latest release version of Marlin is recommended, and the current `bugfix-1.1.x` branch is pretty good overall.

m-dan-7, 2018-02-26:
Thank you for clarifying!

closed 2018-02-26 by m-dan-7