MarlinFirmware/Marlin #9534: Printer not printing from the computer

FAB1150, 2018-02-08:
### Bug Report
- Description: 
hi, I recently got an anet A8, and i flashed marlin on it yesterday. I usually use repetier host to print, and with the stock firmware It worked flawlessly, but when using marlin for every command that is being sent (a part of the heating of the nozzle/bed and the homing) it sends back an echo saying "unknown command: [and the command sent]". 
Note that this happens only when it is reading gcode, not when i move it manually, and it works when printing via USB.
I'm new to marlin, so I haven't edited the configuration.h I found in "example configurations", I only enabled safe homing.
- Expected behaviour: when repetier sends a command to the printer, it should follow it and print.
- Actual behaviour: see this https://youtu.be/3fy1d7GfbKk
- Steps to reproduce:
  - install the firmware to the printer
  - connect repetier host (or another host program, I also tested cura) 
  -start a print

Festivejelly, 2018-02-09:
Have you set repetier up correctly to use Marlin and not Repetier firmware?

MarioStoilov, 2018-02-11:
Could you attach an example sliced file?

thinkyhead, 2018-02-19:
[image]

[image]

FAB1150, 2018-02-26:
Sorry for the silence! I'll try to do that as soon as I'll come back home

boelle, 2019-02-20:
@FAB1150 back home?

FAB1150, 2019-02-20:
@boelle oh well I completely forgot.
I still have the same problem with repetier, I "temporarily" switched to pronterface even though I don't really like it.
I'll update when I come back home, for real this time

boelle, 2019-03-06:
@FAB1150 did you get back home for real this time?

@thinkyhead i think we can close this one

FAB1150, 2019-03-07:
@boelle oh no! I'm at home now, yes. Five minutes (I hope) and I can upload a gcode

FAB1150, 2019-03-07:
here am i from the pc, finally. here is a sliced file (in txt because github doesn't allow .gcode files):

[example.txt](https://github.com/MarlinFirmware/Marlin/files/2941156/example.txt)

thanks for the patience

boelle, 2019-06-22:
@thinkyhead does it make sense to you?

shitcreek, 2019-08-14:
Did you ever get this resolved using Repetier? Or did you end up using Pronterface instead?

FAB1150, 2019-08-15:
No, i have never resolved it. I'm still using pronterface but i really hate it as it's slow and buggy, so i'm still interested in solving this.

shitcreek, 2019-08-16:
From the video, the error code I spotted was 'Error: Line Number is not Last Line Number +1' which could be a Repetier communication issue. The other thing I noticed was that all the unknown commands didn't have any gcode mode preceding them - for example the printer is receiving a command of: X10 Y10 instead of G1 X10 Y10 which further suggests it's an issue with your Repetier setup. You can get around this with 'GCODE_MOTION_MODES' enabled assuming that a gcode is declared however a gcode mode must be declared beforehand.
Could you provide the full error log?

FAB1150, 2019-08-21:
ok, as this was quite some time ago i'll need to do another test print. I'm away for the holidays now, so when i come back i'll do it.

I realized that this sounds like a joke, I swear it isn't!

boelle, 2019-09-24:
@FAB1150 back from holiday and have time to test?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle