MarlinFirmware/Marlin #9535: [bugfix-1.1.x] G21, G90 or M82 resets Z offset (after G28 / M420 L0 S1)

ikarisan, 2018-02-08:
Hi!

I have a strange problem.. again. :)

Normally I use Simplify3D as my preferred slicer. Now I gave Slic3er (Prusa Edition) a try and it seems that the G-Code generated by Slic3r resets the Z offset and the prints start too high.

My Z offset is -0.41mm (set with M851) and in my custom start G-code I print a prime line. This line is printed fine with both slicers! But while S3D's next commands are:

```
G92 E0
G1 E-3.0000 F2700
G1 Z0.200 F480
```

Slic3r is adding these lines:

```
G21
G90 
M82
G92 E0
G1 Z0.200 F7200.000
G1 E-0.20000 F2700.00000
```
 
With the result, that Z is not at 0.2mm but much higher. There is space for two sheets of paper between the nozzle and the bed! So I guess that one of G21, G90 or M82 is resetting the Z offset. 

I have attached the two gcode files.
[cube25x25x2_slic3r.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1706831/cube25x25x2_slic3r.gcode.txt)
[cube25x25x2_s3d.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1706832/cube25x25x2_s3d.gcode.txt)

Is this a bug or did I miss something important?

AnHardt, 2018-02-08:
But while S3D's next commands are:
```
G92 E0 ; set E to 0
G1 E-3.0000 F2700 ; retract E by 3mm with 2700mm/minute
G1 Z0.200 F480 ; move z to 0.2 with 480mm/minute
```
Slic3r is adding these lines:
```
G21 ; Set Units to Millimeters
G90 ; Set to Absolute Positioning
M82 ; Set extruder to absolute mode
G92 E0 ; set E to 0
G1 Z0.200 F7200.000 ; move z to 0.2 with 7200mm/minute
G1 E-0.20000 F2700.00000 ; retract E by 3mm with 2700mm/minute
```
There is nothing generally suspicious in it - but
the Z-move is ordered to be 7200/480 = 15 times faster. 
If that is above you printers abilities ... and the z-speed is not proper limited ... you may lose steps ... you may mess up the position.

ikarisan, 2018-02-08:
It was my fault. No bug!

closed 2018-02-17 by thinkyhead