MarlinFirmware/Marlin #9536: 2.0.x Anet A6 LCD

JohnnyTheOne, 2018-02-08:
The LCD screen does not work properly in 2.0.x but in 1.1.x it does
[image]
my printer is Anet A6 I use the same `Configuration.h` and `Configuration_adv.h` files for both versions

AnHardt, 2018-02-08:
The picture is typical for a too fast timing caused by:
Not enough waits, (*)
too long cables, (dropping voltage and catching noise)
too low voltage, (slowing down the display)
...

(*)
```
// If you want you can define your own set of delays in Configuration.h
//#define ST7920_DELAY_1 DELAY_0_NOP
//#define ST7920_DELAY_2 DELAY_0_NOP
//#define ST7920_DELAY_3 DELAY_0_NOP
```

JohnnyTheOne, 2018-02-08:
I've tried it and it does not work. 
#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_0_NOP
#define ST7920_DELAY_3 DELAY_0_NOP
en
#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_2_NOP
#define ST7920_DELAY_3 DELAY_4_NOP

but why does it work for 1.1.x the same printer?

JohnnyTheOne, 2018-02-10:
#9255 solved this problem.
Is there better way to fix instead modify 
```
//set optimization so ARDUINO optimizes this file
//#pragma GCC optimize (3)
```
in
`ultralcd_st7920_u8glib_rrd_AVR.cpp`

thinkyhead, 2018-02-14:
Perhaps the whole file doesn't need optimization, but only part.

closed 2018-02-14 by thinkyhead