MarlinFirmware/Marlin #9537: Mesh Bed Levelling Cr10s

marcodallago, 2018-02-08:
Hello! I'm having some issues with mesh bed levelling on my Cr10s

**- Description:** 
I can't understand how mesh bed levelling works. Once I do the bed levelling procedure with the lcd step by step guide on my machine, Grid information does not remain on the machine after reboot. Of course EEPROM has to be enabled like i did, and everytime I load settings in bed levelling menu.

**- Expected behaviour:** Should load at startup eeprom settings from mesh bed levelling.

**- Actual behaviour:** After double checking M503 it says that z height value is 0.

**- Steps to reproduce:**
  - Open up machine
  - Bed Level
  - Save settings.
  - Turn Off machine
  - Load settings from eeprom.
  - double check if settings are stored
  - No grid points...

_Shouldn't load automatically eeprom settings at every boot?_

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1707503/config.zip)

Festivejelly, 2018-02-09:
How are you saving it? 
You should hit M500 after leveling to save it to EEPROM

lrpirlet, 2018-02-12:
Try M420 to verify if enabled... enable it with `M420 S1`
ref: http://marlinfw.org/docs/gcode/G029-mbl.html

closed 2018-02-19 by thinkyhead