MarlinFirmware/Marlin #9542: Probing inaccurate while bed is hot

Festivejelly, 2018-02-09:
### Bug Report
- Description: Im not entirely sure this is a code bug or not, but im getting inaccurate bed readings while probing while the bed is hot. I have #define PROBING_HEATERS_OFF un commented and im running marlin 1.1.8 using a servo and microswitch.
- Expected behaviour: When probing the bed hot I should get the same or roughly the same standard deviation as when probing cold
- Actual behaviour: Standard deviation is 5 - 8 times higher.
- Steps to reproduce:
  - Do repeatability test while bed is cold (achieve around 0.001 standard deviation)
 - Heat up bed to 60c and do repeat ability test again now receiving standard deviation between 0.005 and 0.01

I dont understand why this might be especially as the heaters are turned off between probing, so there should be no EMF interference.
Whats odd is probing the bed cold consistently gets me a standard deviation of between 0.001 and 0.002, which in my opinion rules out a mechanical issues.

My thoughts were are the heaters actually being disabled for long enough or are they being turned on too quick or maybe they arnt being turned off quick enough?

I also find it odd that the EMF would affect a microswitch as its mechanical.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1709575/Configs.zip)

ManuelMcLure, 2018-02-09:
Could this be that the microswitch components are expanding as they heat up? It might make sense to do the following test:

- Heat up the bed
- Move the probe into position and just before it triggers
- Wait a while (1/2 hour or so) for the temperature of the probe to stabilize
- Test repeatability again.

Festivejelly, 2018-02-09:
I doubt it, its a plastic switch and probes for a short amount of time so I doubt its changing its temp that much.

jasoroony, 2018-02-10:
My 20cm Z-Axis metal ACME screw warms slowly over a 45 minute period and expands (since I enclosed my machine) when using the heated bed, that means I need to adjust the level on repeated jobs for the first layer to be reliable (I.E. allow the first layer to extrude properly instead of being blocked).  The bed also warps as it gets hot (I.E. bends so the center becomes higher than the edges) so I've got to laugh when reading probe results.  I generally don't want to wait 45 minutes for a stable reading, so don't bother probing the bed.  That's just me though.

Festivejelly, 2018-02-10:
Okay i really cant put my finger on what this is. Its accurate cold but not hot... any ideas? Its a plastic end hitting the bed so it cant be warming up that much. PROBING_HEATERS_OFF definiately helps accuracy. I get about 0.008 standard deviation with it disabled and 0.005 with it enabled. But without the bed on 0.001.

I know my probe is still probably accurate enough even with loss of accuracy but id really like to know what could be causing it because for less accurate probes it makes a difference. For my BL touch as an example, it becomes too inaccurate to use when the bed is on.

Festivejelly, 2018-02-10:
Right I think i know whats happening. With the heaters disabled the bed is slowly cooling down over the course of the probing so the difference becomes larger and larger. What It really should do is wait to go back to target heat before it probes again. Is that possible?

18:21:59.705: 1 of 10: z: 0.000 mean: 0.0000 sigma: 0.000000 min: 0.000 max: 0.000 range: 0.000
18:22:12.252: 2 of 10: z: 0.005 mean: 0.0025 sigma: 0.002500 min: 0.000 max: 0.005 range: 0.005
18:22:24.779: 3 of 10: z: 0.002 mean: 0.0025 sigma: 0.002041 min: 0.000 max: 0.005 range: 0.005
18:22:37.305: 4 of 10: z: 0.005 mean: 0.0031 sigma: 0.002073 min: 0.000 max: 0.005 range: 0.005
18:22:49.828: 5 of 10: z: 0.010 mean: 0.0045 sigma: 0.003317 min: 0.000 max: 0.010 range: 0.010
18:23:02.375: 6 of 10: z: 0.012 mean: 0.0058 sigma: 0.004249 min: 0.000 max: 0.012 range: 0.012
18:23:14.906: 7 of 10: z: 0.012 mean: 0.0068 sigma: 0.004574 min: 0.000 max: 0.012 range: 0.012
18:23:27.433: 8 of 10: z: 0.012 mean: 0.0075 sigma: 0.004677 min: 0.000 max: 0.012 range: 0.012
18:23:39.960: 9 of 10: z: 0.015 mean: 0.0083 sigma: 0.005000 min: 0.000 max: 0.015 range: 0.015
18:23:52.481: 10 of 10: z: 0.017 mean: 0.0092 sigma: 0.005483 min: 0.000 max: 0.017 range: 0.017
18:23:52.784: Finished!
18:23:52.788: Mean: 0.009250 Min: 0.000 Max: 0.017 Range: 0.017
18:23:52.788: Standard Deviation: 0.005483

Festivejelly, 2018-02-10:
I've fixed this issue in my PR: https://github.com/MarlinFirmware/Marlin/pull/9577

closed 2018-02-11 by Festivejelly