MarlinFirmware/Marlin #9547: MKS SBASE 1.3 and optical endstop, needs pull-down resistor

manutenfruits, 2018-02-09:
### Bug Report
- Description: Using branch `bugfix-2.0.x` on a MKS SBASE 1.3, optical endstops will stay TRIGGERED, even after disabling `ENDSTOPPULLUPS`. I've used this board before with smoothieware and the same issue happened with pull-ups and no resistor, **only worked on pull-down**. However, I don't see that option in Marlin.
- Expected behaviour: `M119` will display `TRIGGERED` when triggered, `open` when let go.
- Actual behaviour: `M119` displays `open` at startup, `TRIGGERED` once triggered but won't go back to `open`.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

cjsoong, 2018-02-10:
@manutenfruits 
The MPU should have no function of PULL_LO resistor.
You need to add a pull_lo resistor to the ENDSTOP PCB.

manutenfruits, 2018-02-10:
I checked the voltage and it is floating. I tried switching to 3.3v logic instead of 5v and yes, it still floats, but works :/

thinkyhead, 2018-02-18:
True. Pull _down_ is going to be a whole new option for these boards. We've never had to bother with it before, and Arduino's API only provides for `INPUT_PULLUP`. With LPC176x there's a `gpio_mode` function to set the pin mode, so we'll have to add support for it, applying `PINSEL_PINMODE_PULLDOWN` to the pin(s) in question.

thinkyhead, 2018-02-18:
See if the PR linked above helps.

manutenfruits, 2018-02-27:
It works, thanks a lot for the quick response!

closed 2018-02-27 by manutenfruits