MarlinFirmware/Marlin #9557: At high speeds, print head skips steps with linear advance enabled.

alexyu132, 2018-02-10:
### Bug Report
- Description: 
    When I enable linear advance on my CoreXY printer, stepper skipping occurs occasionally during high speed (~250mm/s) printing. When skipping occurs, the printer makes a loud knock as if the X/Y steppers are instantly stopping and starting again. Without linear advance enabled, this does not occur. Running 4000mm/s^2 acceleration with 15 jerk. Using 1/16 microstepping A4988s for X/Y/Z axes and 1/128 microstepping LV8729 for extruder. Running Marlin 2.0.0-bugfix on an AZSMZ mini. Am I running into step rate limitations at this speed?

- Steps to reproduce:
  - Enable linear advance (K term is currently set to 150)
  - Print a test cube at high speed - skipping starts occurring at the top layers.

Picture:
[image]

Configuration:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1712997/configs.zip)

Sebastianv650, 2018-02-11:
Speed is not the big problem with a pressure advance like LIN_ADVANCE. But acceleration is. You might have a look at #9379.

But in one sentence, pressure adaption with your settings will never work. Pressure changes in real world need time to happen. With such acceleration rates, you have not even close enough time for the changes to happen. I know I have to write a much longer documentation when I'm ready to merge my LIN_ADVANCE v1.5.

thinkyhead, 2018-02-19:
@alexyu132 Let's move discussion and testing to #9700.

closed 2018-02-19 by thinkyhead