MarlinFirmware/Marlin #9558: Inductive sensor bed leveling changes or ignores my Z offset.

opy001, 2018-02-10:
### Bug Report
- Description: Z offset is either ignored or not working as programmed

- Expected behaviour: Auto bed leveling happens and I expect the filament to be perfect on the bed

- Actual behaviour: I set my nozzle via the paper test (everything heated up) at X110 Y110 then did a G92 Z0 then I did a G30 X110 Y110 and got an offset of Z0.25.  I set the offset with M851 Z-0.25 and M500.  I then did a print and the filament didn't stick.  I started over and got a Z offset of 0.77 so I set that and then the nozzle scraped the bed.  I manually set the offset in the LCD raising the nozzle by 1-3 100ths at a time until I got to -0.54 and got my first perfect print since installing my auto leveling switch.  So I decided to do another print without changing a thing (I did store setting in the LDC menu and my offset was still there) and as soon as it started the nozzle put a giant scar on my bed.  I changed the Z offset to -0.20 but it did not change a thing.  I rebooted the printer and restarted Cura and set the Z offset to 0 and then it was not scraping the bed, but the filament was just being dragged into a big mess because it was not close enough.  I have slowly been lowering the offset by -0.05 each time until I got to about -0.41 and it finally printed good again, not perfect but good.  Probably if I go back to about -0.50 where I was close to the first time it would be better.  Has this happened to anyone else and if so how do I fix it?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1713083/Configuration.zip)

- Steps to reproduce:
I have no clue how to reproduce this.

opy001, 2018-02-10:
This must have been a glitch or maybe something with the mainboard because after redoing everything again and finally getting back to -0.50 offset I have done several print jobs and it has not failed.  The only thing I changed, after the second good print, was removing G29 from my start gcode and putting M420 S1 to grab the stored calibration.  It just took too long to calibrate every time I printed something. 
 Maybe on larger jobs I'll consider it, but small stuff doesn't use most of the bed anyway.  I will have to figure out how often I need to calibrate and save, but for now it seems to be working.

opy001, 2018-02-11:
I had to change my offset again because it was scraping the bed.  I had printed multiple items and they all turned out great but this afternoon I started another print and it was scraping to the point no filament would come out.

I really don't want to think it's a sensor issue.

opy001, 2018-02-12:
It did it again.  I am printing everything from the SD card and not plugged into the computer.  I did flash an updated config h file to enable safe z homing, but nothing else changed as far as the offset or calibration.  I checked and the offset was still the same as before I uploaded the firmware, but the nozzle scraped the bed.  I rebooted the printer and it did the same thing.  I changed the offset by a couple of hundredths then saved the changes and changed it back then saved again and that seemed to do the trick.  I'm very very confused and concerned at this point.

revilor, 2018-02-13:
Some heated beds interfere with inductive probes. 

#define PROBING_HEATERS_OFF

could help to get more consistent probe readings.

opy001, 2018-02-13:
Thanks for the suggestion and help!  I'll try it after this print job is complete tomorrow.

So during calibration my bed heat needs to be off.  I guess since I am not setting nozzle height and all this is doing is seeing the height difference across the bed this should not matter.  Is that correct?

Are there any steps I need to do besides probe when cold to make this more reliable?
How often should I calibrate?  Before each print or weekly or monthly?  Not sure how much my bed can change from print to print, but I'm a newb.

ManuelMcLure, 2018-02-13:
Not exactly. It's better to probe the bed when it's hot, since the bed can change shape from cold to hot. What PROBING_HEATERS_OFF does is temporarily turn off the heater just as the probe is being done and immediately turns on the heater again when moving to the next point to probe.

My startup gcode looks like this:

M190 S65 ; Heat bed to 65C
M400 ; Make sure we sync with the above command
G4 S30 ; Wait 30 seconds for bed temp to stabilize
G28 ;Home
G29 ;Bed Level
G0 F3000 X40 Y-5 Z1 ; Somewhere ooze won't matter
M109 S210 ; Heat nozzle to 210C
...

I'm using bilinear leveling with a 3x3 grid and I level before each print. I figure that the time taken to do the leveling is negligible compared to the time it actually takes to do a print (plus the time it takes for the bed to cool down at the end of the print so I can take my parts off the PrintBite+ bed surface).

opy001, 2018-02-13:
Now that makes a lot more sense.  Thank you so much!  It's great to have someplace that can help solve problems and learn at the same time :) .  Still printing from yesterday, but I'll post an update on my findings.

Also, thanks for the gcode example ;)

ManuelMcLure, 2018-02-13:
Note that you don't really want to hard code the temperatures in your slicer's start gcode. Instead use the slicer provided placeholders. For Cura:

M190 S{print_bed_temperature_layer_0}
M109 S{print_temperature_layer_0}

and for Slic3r:

M190 S[first_layer_temperature_0]
M109 S[first_layer_bed_temperature]

opy001, 2018-02-13:
Yep, I let Cura set the temps depending on the profile and filament I use.  Thanks again for your help.

opy001, 2018-02-14:
Well in the initial test the heaters turned off while probing and turn on if needed when moving to the next spot, so that's a plus.  But again, as soon as I started printing the nozzle dug into my bed.  I changed the offset from -0.41 to -0.38 and rebooted and it still dug into my bed.  I changed the offset again to -0.25 and that 0.16 mm did not make one bit of difference.  I then set the offset to 0 and this time the bed heater stayed on at least two of the times the probe went down and came back up (not a huge deal I guess), but it printed somewhat acceptable.

My final test was after I did some changes in the probe offset from extruder settings in the configuration h file.  I set the x and y according to what I just measured, but what I think really helped was setting the Z_PROBE_OFFSET_FROM_EXTRUDER to 2.  My raft and skirts have not looked that good since before I got the probe.  I just started a 6 hour job and all looks perfect.  I'll test again tomorrow.

Again, thanks for your help.  There are so many little things that can affect the outcome that it'll take time to smooth then all out.  But all in all, I love the printer and firmware.

ManuelMcLure, 2018-02-14:
How were you setting offset before you set Z_PROBE_OFFSET_FROM_EXTRUDER? The way I do it is to use the M851 gcode in my startup gcode _before_ I do anything else. M851 should have exactly the same effect as setting Z_PROBE_OFFSET_FROM_EXTRUDER and is easier to change on-the-fly. I've never had much success using the slicer's Z offset parameter.

opy001, 2018-02-14:
I used M581 to begin with and it did the same as going into the lcd and setting it there.  That’s what I’ve been doing this whole time.  Not sure if uploading the new firmware did something but the Offset always stayed after I uploaded the firmware, but everything would act different. This is the first time it’s actually acted normal by setting the offset in the firmware before uploading it. Also, when I set the offset in the firmware this last time it does not reflect that when I check it on the LCD screen.

opy001, 2018-02-14:
I have not used the slicers z offset, just the z offset in the printer itself.

ManuelMcLure, 2018-02-14:
Did you do the M502 M500 M501 dance? I.e. load firmware defaults, save them to EEPROM, and reread the EEPROM? You should do that every time you upload a new firmware configuration to ensure that old configuration settings in your EEPROM don't override the settings from configuration.h/configuration_adv.h.

opy001, 2018-02-14:
Thanks, I will do that the next time I upload.  As for now, it seems to be working.  But it usually works for 3-4 prints then gets totally jacked after that.  But I think the probe with heater off as well as setting offsets on the firmware helped more then anything.

opy001, 2018-02-15:
Just an update, so far so good!  I only had to start one print over and that was because I screwed it up lol.  Thanks for all the help!  My base layers are the best they've ever been now!

closed 2018-02-23 by thinkyhead