MarlinFirmware/Marlin #9560: [bugfix-1.1.x] SD insertion or removal resets printer

lemmingDev, 2018-02-10:
### Bug Report
- Description: SD insertion or removal resets printer
- Expected behaviour: printer should not restart - 
- Actual behaviour: printer resets on SD insertion or removal
- Steps to reproduce:
  - Insert or remove an SD
  - Watch the printer restart

Have MKS Gen 1.4
I have TMC2130 setup through SPI
I have the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled
1.1.8 firmware works fine
Bugfix from a week or two ago is ok
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1713162/Marlin.zip)

Bugfix from today 10/02/18 causes printer to reset
I think it might have to do with changes to SPI stuff in Configuration_adv.h and pins_RAMPS.h

cjsoong, 2018-02-10:
Maybe this problem? 
Marlin 2.0 2018-02-10 also have the same problem
https://github.com/MarlinFirmware/Marlin/pull/9546

lemmingDev, 2018-02-10:
Yep - that was the problem

I reverted back ultralcd.cpp to just before that change was made and everything works fine.

ImplementOfWar, 2018-02-11:
Can confirm this issue.

thinkyhead, 2018-02-19:
What is your `LCD_LANGUAGE` setting?

closed 2018-02-26 by thinkyhead