MarlinFirmware/Marlin #9561: Probed Height Increments with repeated UBL G29 J (Mesh being double compensated?)

dagnall53, 2018-02-10:
### Bug Report 
I am using 1.1.8, recently downloaded, on a XL delta.

I am seeing a continual "creep" in the bed height measurement when using UBL G29 J.
What I would expect is that a "G28 G29 J" should result in consistent readings of my three points on the bed. 

What I am seeing is a consistent increment in the "read bed heights", and offsets in the mesh.  I suspect it may be related to other "height" issues". 

As I understand it, the selected mesh is tilted and height modified by the J probing, and then used for subsequent operation, but I would not have expected the "corrected" mesh to be saved to the eeprom unless explicitly by a G29 S operation?
 
I am using a delta, (and have made mods to the mesh min max as per some of the recent issues), and the mesh its self seems to work fine (repeatedly), (ie no bed height increments with a G28 G29 P0 G29 P1 sequence), as would be expected since the mesh is zeroed at each start.

What I have tested is repeated G29 and G28 G29 J which give the results below. 
I started with a blank mesh and repeated G29 J which gave no "increment behaviors". even without G28 before each G29.

Then I changed to a "filled in" mesh and you can see that initially the bed measurements start at small values similar to those seen in the null mesh values, but they then increment with each G28-G29 J until the bed is apparently nearly 2mm up in the air.. Looking at G29 T1 later showed that the Mesh was being "height increased". 

I would have expected a G29 J on a bed that has just been probed to NOT tilt it as the bed mesh should already reflect the tilt.  I suspect G29 J is making its own reading of the "tilt" and height and "compensating" without taking account of the pre-existing "mesh" tilt and height offsets. 

It should be possible to accommodate this behavior if the G29 J does NOT automatically save the mesh. 

So: What is the intended sequence for using a G29 J at the start of every print? How is it intended to load a (probed) mesh and ensure it is not stored in an altered form for the next print?

Secondly, is it possible to incorporate a "Mesh tilt and average height" measure into the J measurement code, so "existing" tilt and height offsets are not "double compensated" if the J measures similar heights and tilts? 
 
I do wonder if this is similar to the reported height increasing issues, #9344 but it seems subtly different.  

Code below shows what happened during my tests.., but I suspect the real issue is that G29 J is "double compensating" for preexisting mesh tilts and average height offsets.  

`>>> G29 T1
SENDING:G29 T1
Bed Topography Report for CSV:
NAN,NAN,NAN,NAN,NAN,NAN,NAN
NAN,NAN,NAN,NAN,NAN,NAN,NAN
NAN,NAN,NAN,NAN,NAN,NAN,NAN
NAN,NAN,NAN,NAN,NAN,NAN,NAN
NAN,NAN,NAN,NAN,NAN,NAN,NAN
NAN,NAN,NAN,NAN,NAN,NAN,NAN
NAN,NAN,NAN,NAN,NAN,NAN,NAN
>>> G29 J V4
SENDING:G29 J V4
Bed X: 67.175 Y: 67.175 Z: 0.062
Bed X: -91.763 Y: 24.588 Z: 0.248
Bed X: 24.588 Y: -91.763 Z: -0.033
bed plane normal = [0.0014348,-0.0009816,0.9999985]
rotation matrix:+0.999999 +0.000000 -0.001435
+0.000001 +1.000000 +0.000982
+0.001435 -0.000982 +0.999999
D constant: 0.0924826
>>> G29 J V4
SENDING:G29 J V4
Bed X: 67.175 Y: 67.175 Z: 0.054
Bed X: -91.763 Y: 24.588 Z: 0.236
Bed X: 24.588 Y: -91.763 Z: -0.045
bed plane normal = [0.0014151,-0.0010013,0.9999985]
rotation matrix:+0.999999 +0.000000 -0.001415
+0.000001 +1.000000 +0.001001
+0.001415 -0.001001 +0.999999
D constant: 0.0818259
>>> G29 J V4
SENDING:G29 J V4
Bed X: 67.175 Y: 67.175 Z: 0.082
Bed X: -91.763 Y: 24.588 Z: 0.266
Bed X: 24.588 Y: -91.763 Z: -0.009
bed plane normal = [0.0014141,-0.0009507,0.9999985]
rotation matrix:+0.999999 +0.000000 -0.001414
+0.000001 +1.000000 +0.000951
+0.001414 -0.000951 +0.999999
D constant: 0.1131551

Mesh loaded from slot 0
Done.
>>> G29 J V4
SENDING:G29 J V4
Bed X: 67.175 Y: 67.175 Z: 0.085
Bed X: -91.763 Y: 24.588 Z: 0.171
Bed X: 24.588 Y: -91.763 Z: -0.040
bed plane normal = [0.0001487,-0.0008442,0.9999998]
rotation matrix:+1.000000 +0.000000 -0.000149
+0.000000 +1.000000 +0.000844
+0.000149 -0.000844 +1.000000
D constant: 0.0387606
>>> G29 J V4
SENDING:G29 J V4
Bed X: 67.175 Y: 67.175 Z: 0.119
Bed X: -91.763 Y: 24.588 Z: 0.118
Bed X: 24.588 Y: -91.763 Z: 0.086
bed plane normal = [-0.0014577,0.0009124,0.9999985]
rotation matrix:+0.999999 +0.000000 +0.001458
+0.000001 +1.000000 -0.000912
-0.001458 +0.000912 +0.999999
D constant: 0.0822211
>>>    G28
SENDING:G28
>>> G29 J V4
SENDING:G29 J V4
Bed X: 67.175 Y: 67.175 Z: 0.530
Bed X: -91.763 Y: 24.588 Z: 0.508
Bed X: 24.588 Y: -91.763 Z: 0.368
bed plane normal = [-0.0003661,0.0017746,0.9999983]
rotation matrix:+1.000000 +0.000000 +0.000366
+0.000001 +0.999998 -0.001775
-0.000366 +0.001775 +0.999998
D constant: 0.6249018
>>>    G28
SENDING:G28
>>> G29 J V4
SENDING:G29 J V4
Bed X: 67.175 Y: 67.175 Z: 1.785
Bed X: -91.763 Y: 24.588 Z: 1.394
Bed X: 24.588 Y: -91.763 Z: 1.372
bed plane normal = [0.0010205,0.0053798,0.9999850]
rotation matrix:+1.000000 +0.000000 -0.001021
-0.000005 +0.999985 -0.005380
+0.001021 +0.005380 +0.999985
D constant: 2.2144265`

-
Thanks
Dagnall

dagnall53, 2018-02-15:
@thinkyhead @Roxy-3D  Sorry to poke you on this.
Have I misunderstood the purpose of UBL "G29 J". 
If done with no previous mesh, it does a nice 3 point level and height adjust. (as expected)
If done with a non zero mesh, it appears to try to correct the mesh, but because it is using the "new" 3 point probe depths (not taking into account previous mesh probes), it effectively double correctsheight and tilt offsets.

Given that any multi probed mesh is already trying to correct bed height, tilt ,concavity, surface , etc, It appears to me that a G29 J should be probing **using** the (current) mesh offsets. 
If the 3 probes then read the same as the multi probed mesh measurements at the probe points, then any tilt (etc) change will be zero. If there are "no" mesh offsets or the mesh is null, the result will still give a simple tilt and height corrected mesh. 

The reason this is important is that currently I (and others I assume) have to do a G29 P1 before prints, which can be tedious. Using  a G29 J to correct any minor tilt / height offset (but retaining the basic Multi probe bed "shape") is much faster. 

- Again ,sorry to poke you on this, but I got no comments from the issue as presented, so I presume it did not get noticed.

Roxy-3D, 2018-02-15:
> Given that any multi probed mesh is already trying to correct bed height, tilt ,concavity, surface , etc, It appears to me that a G29 J should be probing using the (current) mesh offsets.

I agree...   I think the probing for G29 J should be using the mesh points to correct the probed values to be used to generate the mesh tilt.

thinkyhead, 2018-02-23:
Closing since the discussion at #9622 is converging on the solution. Please continue there.

closed 2018-02-23 by thinkyhead