MarlinFirmware/Marlin #9564: Can we fix the bltouch EMI problem with an inline EMI filter?

dasflux, 2018-02-10:
Not sure if asking questions like this fits the format here.  But you guys are literally the only ones that really talk or know anything about the problem.

Im from the tarantula community.  Like the title says, would one of these filters plus sheilding the wires fix this problem?  Im not circuit savvy enough to know which one of these to buy, but mouser has a huge list  I use the heaters off function in marlin which has helped like many report but not solved all the weirdness.

Roxy-3D, 2018-02-10:
It would depend on where the electrical noise is coming from and where the filter is placed.    Usually, the BL-Touch wires are in the same bundle as the nozzle heater and extruder stepper motor wires.   There is a lot of electrical noise that can be picked up.   Putting a filter any where except right at the BL-Touch probably won't help.     A large capacitor (500 uf ????) physically close to the BL-Touch would probably help a lot if you have electrical noise problems with your BL-Touch.

dasflux, 2018-02-10:
I forgot to mention in combination eith shielding the wires?  And hell I guess wouldnt that mean shielding the unit too?  Man this problem is a pain in the ass.  So glad the piezo probe is around.

Roxy-3D, 2018-02-10:
Personally...   I think a servo that kicks down a probe leg with a micro-switch on the end of it is the most trouble free way to have a Z-Probe.

dasflux, 2018-02-10:
I actually have that one posted on r/tevotarantula actually.  The sg90 =) .  I doing a rebuild and haven't for my piezo probe yet, but have you heard any problems with it yet?

Roxy-3D, 2018-02-10:
This is definitely opinion and there will be many that disagree.   But the inductive, capacitive and piezo probes are just asking for trouble.   When they work, they work well.   But so many people have problems with them.  Especially when they are trying to get them working initially.    The BL-Touch is a nice unit but it has some strange behavior that is difficult to work around some times.

I like servo's with a probe leg!   They are simple.   They are easy to get working.   And they are not very sensitive to electrical noise.

studiodyne, 2018-02-10:
Suggestions
1-) use Ups , because the more part of noise use power wires
2-) Use 12v or  24v battery just for probing , to know if a good power supply can help
3-) make a faraday box with aluminium paper
4-) Connect your buildplate  or all that is metallic to the ground
5-) Test the ground and equipotential  ' to test the ground with the special device , go on google you will have some tips to make it

If you are under a  4G antenna that pulse 10kw of power , it's sure , the last solution is go elsewhere

dasflux, 2018-02-10:
I think im gonna give the servo probe a chance.  I see the sg90 and mg90, which would you recommend?

Roxy-3D, 2018-02-10:
I've mostly used the SG90 servo's.    But the MG90 servo's are higher quality with metal gears internally.   This might not be true...   But I'm under the impression that MG90 servo's are less likely to jitter.

closed 2018-02-19 by thinkyhead