MarlinFirmware/Marlin #9567: upgraded to Marlin 1.1 used with hictop MKS board dual extruder SD card no longer works

kpdelucia, 2018-02-10:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: --I upgraded to marlin 1.1 posted by gameofthefuture on thingiverse to use MKS board not ramps generic. It enable my second extrude head properly but now it doesnot see my sdcard in my repwrap discount smart controller-
- Expected behaviour: --The SD card reader worked previously-
- Actual behaviour: --sd card inserted message comes up   but when i navigate to print from SD " it says "NO SD CARD"-
- Steps to reproduce:

  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```
[Configurations for gitpost.zip](https://github.com/MarlinFirmware/Marlin/files/1713662/Configurations.for.gitpost.zip)

thinkyhead, 2018-02-19:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@kpdelucia Please test with the latest `bugfix-2.0.x`

kpdelucia, 2019-02-20:
Ok i have had many problems using using newer versions of Arduino with that printer.  But I will give it a try this weekend. Using latest Arduino and latest Marlin

boelle, 2019-03-06:
@kpdelucia what results did you get that weekend?

@thinkyhead i think we can close this one

kpdelucia, 2019-03-06:
Sorry to hold you up I had to print a $$ job on that printer last weekend with a single extruder.  I will try this weekend but if you are really anxious to close this out, Don't wait for me. I will likely figure it out.

boelle, 2019-03-06:
no no, we can wait a few more days :-D

kpdelucia, 2019-03-13:
tried latest marlin but I am having a hard time setting steps per mm for each extruder.
  //#define DISTINCT_E_FACTORS                             //   *****************kirk look************

/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 200, 200, 1200, 180 }    // z was 1159

If I put a number after the 180,  I get a compile error about too many arguments.  I suspect I have to uncomment the "define distinct" above it ...
update.  Did that now it compiles and uploads.

boelle, 2019-03-13:
attach 2 config files as zip

thinkyhead, 2019-03-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-03-13 by thinkyhead

kpdelucia, 2019-03-14:
Thanks @boelle I got it mostly working now but I don't want to discuss it in the wrong place so Thanks for showing me the bug was fixed for the sd card not found