MarlinFirmware/Marlin #9568: UBL strange behavior: Outside of bedsize and over fadeheight

studiodyne, 2018-02-10:
I have a bucket purge outside of bed , when ubl activated , and in the faded height ,  when i go outside of the bed , the carriage stop at bedsize and up the carriage infinitly ' 

1-) Is it normal  or bug or need support?
2-) Is it fixed now  ?
3-) Is there a ' out of boundaries ubl normal behavior ' option ?
4-) Is there a topic about this issue

BUT MY ONLY IMPORTANT QUESTION IS 👍 
Can i use G29 D ' disable ' , during a print to go outside of the bed , and G29A 'enable ' when back to the print ?

Thanks

But i

closed 2018-02-11 by studiodyne

Roxy-3D, 2018-02-11:
Yes...   You can do Idea 5 to work around the issue.    But I'm going to find this issue and get it fixed!!!

thinkyhead, 2018-02-14:
> 'that speaks french now loll '

I studied French for two years in High School. But I never kept it up, so now, while I can generally comprehend French, I cannot write or speak it with any fluency.

Not all of us Americans are 100% provincial hicks.