MarlinFirmware/Marlin #9569: printer halts after inserting sd card

zrhck, 2018-02-10:
- Description: Printer halts after putting in a sd card into RepRapDiscount Smart Controller
- Expected behaviour: initialize sd card
- Actual behaviour: Printer halts and i have to reset it. Error from S3D: 
Recv: sd card Error:KILL caused by KILL button/pin

- Steps to reproduce:
  - Insert SD card or start with sd card inside RAMPS RepRapDiscount Smart Controller

[configuration_h_files.zip](https://github.com/MarlinFirmware/Marlin/files/1713687/configuration_h_files.zip)

studiodyne, 2018-02-10:
it's a bad pin configuration
Very all pins attributions , ask your reseller , ask all that have the same printer here
You will gain time if you place here  the name of your printer , the board , and your pins config

zrhck, 2018-02-10:
Thank you for your reply @studiodyne 
It's an Anet A8 with RAMPS 1.4. I just checked the matching wiki [post](http://reprap.org/wiki/RepRapDiscount_Smart_Controller)

[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/1713702/pins_RAMPS.zip)

closed 2018-02-10 by zrhck