MarlinFirmware/Marlin #9573: [FR]: Wait for bed heating between probes

Festivejelly, 2018-02-10:
### Feature Request
I've tested this in the Marlin_Cpp already and it works

So if you have heaters disabled between probing the bed can slowly decrease in temp while probing a mesh. If we add the below code it waits between probes until the bed is back up to the desired temp.
It has made my probe MUCH more accurate.

It goes in the do_probe_move method just after this:

    // Deploy BLTouch at the start of any probe
    #if ENABLED(BLTOUCH)
      if (set_bltouch_deployed(true)) return true;
    #endif

```
     do {
        SERIAL_ECHOPAIR("Target Bed", thermalManager.degTargetBed());
        SERIAL_ECHOPAIR("Actual Bed", thermalManager.degBed());
      if(thermalManager.degBed() >= thermalManager.degTargetBed()){
        break;
      }
      safe_delay(1000);
     } while(true);
```

Although the delay seems to suppress all output. I didnt want to do a pull request that was buggy so maybe someone could check this out?

Festivejelly, 2018-02-10:
Pull request here: https://github.com/MarlinFirmware/Marlin/pull/9577

Festivejelly, 2018-02-10:
I just want to say ive tested this quite a bit and it is for sure making my sensor a hell of a lot more accurate during repeatability tests. Which means I think it will also increase accuracy when you're probing a mesh and the bed wont be see sawing so much between probing points.

thinkyhead, 2018-02-19:
Move topic to #9647

closed 2018-02-19 by thinkyhead