MarlinFirmware/Marlin #9574: Z is Negative after G29 Autoleveling

Jeepguy42, 2018-02-10:
### Bug Report
- Description: after Home X Y and Z axis I run a G29 to run the autoleveling before printing. After the G29 runz, Z says its at -37 (+ or - .2mm). This means when it goes to Z0 it prints in the air. I've tried adjusting Z probe offset any it doesn't correct the problem. If I change the distance clearance between probes it does change the -37 by the amount I change it.
- Expected behaviour: Home Z and then running G29 should level the bed but after leveling the probe should sit at roughly at Z7 (my probe offset amount)
- Actual behaviour: after auto leveling the Z thyingks its @ -37

I'm new to this but their aren't really logs on what to do when starting from scratch.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Marlin Files.zip](https://github.com/MarlinFirmware/Marlin/files/1713770/Marlin.Files.zip)

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

MarioStoilov, 2018-02-11:
Is the printer connected to a PC or anything that can read the serial output? Usually, the after the G29 code, you get the grid/plane that the printer will use on the serial output.

Jeepguy42, 2018-02-12:
I print with this printer using Repetier. It is connected via USB.

AnHardt, 2018-02-12:
What is at -37?
The nozzle itselves? Unlikely to be below the bed.  
Marlins display?
Repetier Hosts display?

If just RH try a version before 2.0..0 They changed the way to interpret our position feedbacks.

MarioStoilov, 2018-02-12:
If you are printing with repetier host, turn on the logs and and give us your output after g29.

AnHardt, 2018-02-12:
> Expected behaviour: Home Z and then running G29 should level the bed but after leveling the probe should sit at roughly at Z7 (my probe offset amount)

This expectation is wrong.
The nozzle will end at max( Z_CLEARANCE_DEPLOY_PROBE, Z_CLEARANCE_BETWEEN_PROBES).
So the 'trigger point' of the probe z-probe-offset above that. (z-probe-offset is usually negative - so the trigger point will still be below the nozzle but above the bed. ) Mechanical probes have now to be stowed to print. (Fix mounted probes are simply switched off (signal ignored). The bottom surface of inductive/capacitive/IR/...  working_on_distave/not_touching probes is completely irrelevant. What matters is their trigger point.)

Jeepguy42, 2018-02-12:
I will try the autoleveling again tonight with the logs on.

The -37 is where repetier (M114) is reporting the position of the Nozzle so when it starts to print and moved to the 0 Z height its actually prints in the air since its moving up to get to 0. My probe is an optical probe but has a standard deviation of .001 (roughly) when running an M48 repeatability test. My probe  sits above my nozzle but triggers roughly 7mm below the nozzle.

I have my Z_CLEARANCE_DEPLOY_PROBE set to 0 and my Z_CLEARANCE_BETWEEN_PROBES set to 5. adjusting either number doesn't affect the -37 result.

AnHardt, 2018-02-12:
Any number for X,Y,Z you see on the RH display (above V2.0.0) is completely meaningless.

Jeepguy42, 2018-02-12:
Is their a better program to be using to test this error in to check to see what the Z height? I'm only familiar with RH as my only other Printer is a Original Prusa MK2S and I run it through Octoprint without issue.

AnHardt, 2018-02-12:
If you are familiar with RH use a version before 2.0.0. They are still downloadable.

AnHardt, 2018-02-12:
Your Configuration.h
```
...
/**
 * Software Endstops
 *
 * - Prevent moves outside the set machine bounds.
 * - Individual axes can be disabled, if desired.
 * - X and Y only apply to Cartesian robots.
 * - Use 'M211' to set software endstops on/off or report current state
 */

// Min software endstops curtail movement below minimum coordinate bounds
//#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  #define MIN_SOFTWARE_ENDSTOP_Z
#endif
...
```
Your Configuration_adv.h:
```
...
// @section homing

// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
//#define ENDSTOPS_ALWAYS_ON_DEFAULT
...
```
All kinds of endstops are off.

Jeepguy42, 2018-02-13:
I'm using RH version V1.6.1 so I would assume the Z is showing the correct value. I switched to Ponterface as the Repetier log just said Busy: Echoing and gave me no actual log even after running M111 S247

I ran M111 S247 and then ran the G29, This is what the Log reads:

>>>G29
SENDING:G29
echo:G29
  current_position=(0.00, 100.00, 152.70) : >>> G29
Machine Type: Core
Probe: FIX_MOUNTED_PROBE
Probe Offset X:31 Y:22 Z:40.00 (Right-Back & Above Nozzle)
Auto Bed Leveling: LINEAR (disabled)
  current_position=(0.00, 100.00, 152.70) : set_probe_deployed
deploy: 1
do_probe_raise(3.00)
>>> do_blocking_move_to(0.00, 100.00, 152.70)
<<< do_blocking_move_to
  current_position=(0.00, 100.00, 152.70) : setup_for_endstop_or_probe_move
>>> probe_pt(35.00, 25.00, no stow)
  current_position=(0.00, 100.00, 152.70) :
>>> do_blocking_move_to(4.00, 3.00, 152.70)
<<< do_blocking_move_to
  current_position=(4.00, 3.00, 152.70) : set_probe_deployed
deploy: 1
  current_position=(4.00, 3.00, 152.70) : >>> run_z_probe
  current_position=(4.00, 3.00, 152.70) : >>> do_probe_move
>>> do_blocking_move_to(4.00, 3.00, 0.00)
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to
  current_position=(4.00, 3.00, 4.84) : sync_plan_position
  current_position=(4.00, 3.00, 4.84) : <<< do_probe_move
>>> do_blocking_move_to(4.00, 3.00, 7.84)
<<< do_blocking_move_to
  current_position=(4.00, 3.00, 7.84) : >>> do_probe_move
>>> do_blocking_move_to(4.00, 3.00, -10.00)
echo:busy: processing
<<< do_blocking_move_to
  current_position=(4.00, 3.00, 4.86) : sync_plan_position
  current_position=(4.00, 3.00, 4.86) : <<< do_probe_move
>>> do_blocking_move_to(4.00, 3.00, 7.86)
<<< do_blocking_move_to
  current_position=(4.00, 3.00, 7.86) : >>> do_probe_move
>>> do_blocking_move_to(4.00, 3.00, -10.00)
echo:busy: processing
<<< do_blocking_move_to
  current_position=(4.00, 3.00, 4.87) : sync_plan_position
  current_position=(4.00, 3.00, 4.87) : <<< do_probe_move
>>> do_blocking_move_to(4.00, 3.00, 7.87)
<<< do_blocking_move_to
  current_position=(4.00, 3.00, 7.87) : >>> do_probe_move
>>> do_blocking_move_to(4.00, 3.00, -10.00)
echo:busy: processing
[... 391 lines cut ...]
>>> do_blocking_move_to(84.00, 77.00, 8.87)
echo:busy: processing
<<< do_blocking_move_to
  current_position=(84.00, 77.00, 8.87) : >>> do_probe_move
>>> do_blocking_move_to(84.00, 77.00, -10.00)
<<< do_blocking_move_to
  current_position=(84.00, 77.00, 5.87) : sync_plan_position
  current_position=(84.00, 77.00, 5.87) : <<< do_probe_move
>>> do_blocking_move_to(84.00, 77.00, 8.87)
echo:busy: processing
<<< do_blocking_move_to
<<< probe_pt
  current_position=(84.00, 77.00, 8.87) : set_probe_deployed
deploy: 0
do_probe_raise(3.00)
>>> do_blocking_move_to(84.00, 77.00, 8.87)
<<< do_blocking_move_to
  current_position=(84.00, 77.00, 8.87) : > probing complete
  current_position=(84.00, 77.00, 8.87) : G29 uncorrected XYZ
Z from Probe:-37.02  Matrix:8.66  Discrepancy:-45.68
  current_position=(84.04, 77.07, -37.02) : G29 corrected XYZ
  current_position=(84.04, 77.07, -37.02) : clean_up_after_endstop_or_probe_move
<<< G29
  current_position=(84.04, 77.07, -37.02) : sync_plan_position
echo:M105

Jeepguy42, 2018-02-13:
I noticed it says my Probes z position is set to 40. I went into Arduino and checked that I had it set to 7 and I did. I reuploaded it to the ramps board and redid the G29 and it still says the probes z position is 40.

Jeepguy42, 2018-02-13:
Last I checked I have enable EEPROMS as I've saved the steps back using the M92 command and the M500 to save them without issue.

Are you referring to run **G29.1** or **M565**? I believe I've tried **G29.1 Z-7** but it doesn't seam to care and still sees the probe offset as **X31 Y22 Z40**. I can try **M565** and set the Z to -7 (**M565 X31 Y22 Z-7**) and see if that corrects it. Any other thoughts?

Roxy-3D, 2018-02-13:
M502 followed by M500.

Jeepguy42, 2018-02-14:
I did the M565 X31 Y22 Z-7 and it didn't seem to do anything (after running G29 it still showed at -37). I then did the M502 followed by M500 as suggested by Roxy-3D. I then Homed Z and reran the G29 and it said the Nozzle was at +10 instead of -37. I don't know what was hanging it up but it didn't seem to want to correct until I did that. I thought I tried that once but maybe it didn't send properly using Repetier Host.

It was slightly to high so I'm currently trying a different Z-Probe Offset but this seams to have fixed the issue. Any idea why it was holding the settings?

AnHardt, 2018-02-14:
Holding the settings is the wanted function when enabling EEPROM.

Jeepguy42, 2018-02-14:
I getting the holding setting so it doesn't always revert back to defaults but when I was saving back settings such as probe offsets and they weren't holding. seems like something else was interfering.

No biggie I guess, it is working now. It might be something to look into if more people see it happening.

Thanks for the help on this one.

closed 2018-02-14 by Jeepguy42