MarlinFirmware/Marlin #9579: [bugfix-1.1.x] TMC bump sensitivity via M914 not working for Y stepper driver

MoellerDi, 2018-02-11:
Setting TMC bump sensitivity (M914) via terminal only adjust TMC bump sensitivity for the X stepper driver but not for the Y stepper driver (at least for me).

TMC bump sensitivity settings from configuration_adv.h works fine but cannot be changed for the Y driver w/o re-flashing Marlin. Currently I am re-flashing Marlin whenever I need to further change/fine-tune this setting.

Printer: CR-10 + MKS Gen 1.4 + TMC2130 (YXZE)
Firmware: latest Marlin bugfix-1.1.x

- Steps to reproduce:
Get current settings via M503
**Send: M503**
<.. cut ..>
Recv: echo:Stepper driver current:
Recv: echo:  M906 X 750 Y 750 Z 650 E0 650
Recv: echo:Sensorless homing threshold:
Recv: echo:  M914 X40 Y40
<.. cut ..>
Recv: ok

Change TMC bump sensitivity via M914
**Send: M914 X41 Y41**
Recv: X driver homing sensitivity set to 41
Recv: X driver homing sensitivity set to 41
Recv: ok

Get settings via M503 again to confirm changes
**Send: M503**
<.. cut ..>
Recv: echo:Stepper driver current:
Recv: echo:  M906 X 750 Y 750 Z 650 E0 650
Recv: echo:Sensorless homing threshold:
Recv: echo:  M914 X41 **Y40**
<.. cut ..>
Recv: ok

teemuatlut, 2018-02-11:
Can't reproduce.
Does M906 work? Check the actual RMS values with M122 as those are read from the driver and not the library.

MoellerDi, 2018-02-11:
@teemuatlut thank you for your quick action and looking into this. 

Just in case it's needed, my configs/sources can be found on github at https://github.com/MoellerDi/Marlin/tree/bugfix-1.1.x-HD-MKSGen14

I just did a quick check and M906 is working fine:

**Send: M503**
<.. cut ..>
Recv: echo:Stepper driver current:
Recv: echo:  M906 X 750 Y 750 Z 650 E0 650
Recv: echo:Sensorless homing threshold:
Recv: echo:  M914 X40 Y40
<.. cut ..>
Recv: ok
[...]
**Send: M906 X800 Y800 Z700 E0700**
Recv: X axis driver current: 800
Recv: Y axis driver current: 800
Recv: Z axis driver current: 700
Recv: E0 axis driver current: 700
Recv: ok
[...]
**Send: M503**
<.. cut ..>
Recv: echo:Stepper driver current:
Recv: echo:  M906 X 800 Y 800 Z 700 E0 700 ; new values set w/o issues
Recv: echo:Sensorless homing threshold:
Recv: echo:  M914 X40 Y40
<.. cut ..>
Recv: ok

However, it seems Y driver homing sensitivity values get not written to the driver. Below I confirmed this with M122. BTW, for me it looks like M914 does the same thing twice for the X driver, at least its saying "X driver homing sensitivity set to .." two times. So setting Y driver homing sensitivity seems be missing on my printer.

**Send: M122**
Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	750	750	650	650
Recv: RMS current	734	734	642	642
Recv: MAX current	1035	1035	905	905
Recv: Run current	23/31	23/31	20/31	20/31
Recv: Hold current	11/31	11/31	10/31	10/31
Recv: CS actual		11/31	11/31	10/31	10/31
Recv: PWM scale	1	1	1	1
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysterisis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
**Recv: Stallguard thrs	40	40	0	0**
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:0B:00:00
Recv: 	Y = 0x80:0B:00:00
Recv: 	Z = 0x80:0A:00:00
Recv: 	E0 = 0x80:0A:00:00
Recv: 
Recv: 
Recv: ok
[...]
**Send: M914 X41 Y41**
Recv: X driver homing sensitivity set to 41
Recv: X driver homing sensitivity set to 41 ; **shouldn't this be for the Y driver ?**
Recv: ok
[...]
**Send: M122**
Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	750	750	650	650
Recv: RMS current	734	734	642	642
Recv: MAX current	1035	1035	905	905
Recv: Run current	23/31	23/31	20/31	20/31
Recv: Hold current	11/31	11/31	10/31	10/31
Recv: CS actual		11/31	11/31	10/31	10/31
Recv: PWM scale	1	1	1	1
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysterisis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
**Recv: Stallguard thrs	41	40	0	0**
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:0B:00:00
Recv: 	Y = 0x80:0B:00:00
Recv: 	Z = 0x80:0A:00:00
Recv: 	E0 = 0x80:0A:00:00
Recv: 
Recv: 
Recv: ok

teemuatlut, 2018-02-11:
Okey yeah I see the bug now. I'll get a PR ready today.
This can be considered confirmed for 1.1.x. The v2 branch is working correctly.

MoellerDi, 2018-02-11:
@teemuatlut thanks for the quick bugfix. It's now working fine on my printer ;-)

Send: M914 X39 Y39
Recv: X driver homing sensitivity set to 39
Recv: Y driver homing sensitivity set to 39
Recv: ok
[...]
Send: M122
Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	700	700	600	600
Recv: RMS current	673	673	581	581
Recv: MAX current	949	949	819	819
Recv: Run current	21/31	21/31	18/31	18/31
Recv: Hold current	10/31	10/31	9/31	9/31
Recv: CS actual		10/31	10/31	9/31	9/31
Recv: PWM scale	3	3	3	3
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysterisis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs	39	39	0	0
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:0A:00:00
Recv: 	Y = 0x80:0A:00:00
Recv: 	Z = 0x80:09:00:00
Recv: 	E0 = 0x80:09:00:00
Recv: 
Recv: 
Recv: ok
```

closed 2018-02-11 by MoellerDi