MarlinFirmware/Marlin #9583: Bug Report

notsatisfied, 2018-02-11:
### Bug Report
I can run any commands through pronterface, but I cannot get Marlin with ramps 1.4 and an arduino mega to actualy start printing. 

Load obj stl or gcode file into pronterface, press begin printing after slicing is complete through slicer. Printer runs homing and stops, heating is complete and stays hot. does nothing else for hours.

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/1714052/config.files.zip)

Lyr3x, 2018-02-11:
did you tried to print via sd-card?

notsatisfied, 2018-02-12:
I have not as I positioned pieces of my printer in a not so bright way where my bed would clip the sd card in its movement during the print

Lyr3x, 2018-02-12:
Maybe you should try this First to be sure that the problem isnt just pronterface or sth else. 

Also: your issue title isnt very specific. When you open an issue there is a template for bug reports

notsatisfied, 2018-02-12:
sorry about that, I thought thats what it meant in the template.  I just talked to a coworker today who said it may be issues with waiting for the heated bed to reach temperature. I will try this, and if it does not fix the issue I will dismount the lcd and print from sd.

Lyr3x, 2018-02-13:
no problem 👍 also it will be helpful if you provide some more information about your printer :)

thinkyhead, 2018-02-20:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@notsatisfied Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek