MarlinFirmware/Marlin #9584: lcd issues

notsatisfied, 2018-02-11:
I have marlin with ramps 1.4 installed on my custom printer. There are a couple of bugs that are hard to get around though. the main issue is the lcd screen (reprap discount full graphic smart controller) works sometimes but every time i need to use the display it is showing not the menu or anything readable. Instead it randomly will display characters in odd places, I have seen i think Chinese characters as well as total gibberish. occasionally it also will display correctly but offset from the center of the screen so part is cut off and the other side is blank.
[config files.zip](https://github.com/MarlinFirmware/Marlin/files/1714054/config.files.zip)

Bob-the-Kuhn, 2018-02-13:
I'm assuming that you are running one of the Marlin 1.1.xx releases.

I expect that the LCD's SPI interface is being run too fast or is picking up electrical noise.  That display is VERY finicky.  

Try separating the LCD's cables from everything else and see if that cures it.

The next step is to modify the delays in the file **ultralcd_st7920_u8glib_rrd.h**

thinkyhead, 2018-02-23:
This issue will be closed soon due to lack of activity. If you have solved the issue, please let us know how you solved it and close this issue. If you haven't, please tell us what else you've tried in the meantime.

closed 2018-02-28 by thinkyhead