MarlinFirmware/Marlin #9585: [2.0.X]compile failed with MKS SBASE due to port error

MS1987, 2018-02-11:
------------------------------------------------------------
### Compile Error
Use marlin 2.0, when I compiled with MKS SBASE:
          `platformio run -e LPC1768` 
 I got an error:
```
Marlin\src\module\configuration_store.cpp:1866:69: error: 'port' was not declared in this scope
#define CONFIG_ECHO_START do{ if (!forReplay) SERIAL_ECHO_START_P(port); }while(0)
^
Marlin\src\module\../inc/../core/serial.h:107:64: note: in definition of macro 'SERIAL_ECHO_START_P'
#define SERIAL_ECHO_START_P(p)             (serialprintPGM_P(p,echomagic))
^
Marlin\src\module\configuration_store.cpp:1881:5: note: in expansion of macro 'CONFIG_ECHO_START'
CONFIG_ECHO_START;
^~~~~~~~~~~~~~~~~
Marlin\src\module\configuration_store.cpp:1892:26: error: 'port' was not declared in this scope
SERIAL_ECHOLNPGM_P(port, "  G21    ; Units in mm");
^
Marlin\src\module\../inc/../core/serial.h:103:69: note: in definition of macro 'SERIAL_PROTOCOLLNPGM_P'
#define SERIAL_PROTOCOLLNPGM_P(p,x)             (serialprintPGM_P(p,PSTR(x "\n")))
^
Marlin\src\module\configuration_store.cpp:1892:7: note: in expansion of macro 'SERIAL_ECHOLNPGM_P'
SERIAL_ECHOLNPGM_P(port, "  G21    ; Units in mm");
^~~~~~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\module\configuration_store.o] Error 1
```

------------------------------------------------------------
Even I have tried the modification of [#9155](https://github.com/MarlinFirmware/Marlin/issues/9155). Still exists.

```

MS1987, 2018-02-11:
I have checked that it occurs after the commit of [#8148](https://github.com/Spawn32/Marlin/commit/f7efac57b7ac607405529a71b4e36a6545525c89#diff-8d5923b750f73e38d70738067f9e89ff). The previous version works well.

Spawn32, 2018-02-11:
this is strange, i have tested the version from 10.02.18, no compiler error, i have serialport as -1 and serialport_2 as 0, the rest is like the example config in examples / Mks Sbase...

MS1987, 2018-02-12:
I check again, it works! Maybe I have mixed the old version and the new one. Thanks.

closed 2018-02-12 by MS1987