MarlinFirmware/Marlin #9593: Serial disconnecting from host mid print and no control over digipot current, SBASE / LPC1768

Spawn32, 2018-02-11:
Get random disconnects on the USB port, can be after a minute and even 2 hours, haven't been able to finish 1 print.

Only thing changed on my side is that i have changed from a I3 clone to a big corexy..

Also got Firmware unresponsive a few times..

The Digipot current setup also dosent seem to be working, whatever you set them to they act exactly the same..

And yet another bug, tested to turn of the watchdog to see how it behaved, not possible, it throws a compile error without it...

Ended up using a small AVR melzi board with Marlin 1.1.8 to get my prints done 😭

Spawn32, 2018-02-11:
Just remembered, i got a warning that there was a new version of platformio so i updated, could that be the problem ?

Spawn32, 2018-02-11:
Would be very nice if someone with a similar setup or even just a mks sbase, LPC1768 or a Re-Arm could try to verify this faults for me..

briareos77, 2018-02-11:
I have 2 Re-Arm with ramps 1.4 on two of my core xy printers. I get random disconnects also from octoprint and just printing directly from SD card. After the disconnect, everything is frozen.

If i turn off linear advance, it prints all the way through. I have not been able to print a benchy all the way through with linear advance on. I'm not sure if it's related to linear advance, since the disconnects are kind of random.

@Spawn32 Do you have linear advance on also?

Spawn32, 2018-02-11:
I had a very bad Y axes shift when i tested with linear advance enabled, but i found at that linear advance wasn't the problem with that made the shift happen, but have turn it off just to have one less variable until this get sorted, so my problem is not caused but linear advance.

linear advance should / could make problems on slower MCU's like the AVR, but a fast 32bit should be ok...

I have a old (nov 2017) branch of 2.0 bug fix, and that dos not have the usb drops or the digipot problem, but not sure where to look, it's a lott of commits since November and until today .)

briareos77, 2018-02-17:
@Spawn32 Have you been able to isolate the commit that the printer started to freeze at?

I'm using a bltouch and tmc 2130 drivers. What do you have hooked up to your core xy printer?

cdedwards, 2018-04-21:
did you ever get your random disconnects figured out? I'm running into the same situation with a delta and re-arm board. doesn't happen during a print, but randomly while doing other commands. G28, G29, G48 etc.

boelle, 2019-02-20:
@Spawn32 still having issues with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek