MarlinFirmware/Marlin #9596: HELP UBL !

studiodyne, 2018-02-11:
@thinkyhead @Roxy-3D 

Hi , UBL strange behavior outside of bedsize , up infinitly and more strange things, i have a purge bucket outside of the bed ' impossible to probe this void zone ' 

How to solve ?
1-) Enable/disable  UBL by gcode when go out of boundaries " when go on pause zone '
2-) Move to the limit and go on pause zone , to avoid interpolation of bedzone and nobedzone
3-) Add bucket zone in probing points

Thanks

Roxy-3D, 2018-02-11:
There have been a number of reports of the Z-Axis going to infinity on UBL.   And also a few reports of the Z-Axis diving into the bed.     I don't know when the regression got into the code.    I'm going to be making some additions to the UBL code so I'm going to be watching for this.  (AND FIX IT!!!!)

But the problem has to do with the nozzle going off the mesh.    You can do idea 1 and that will work.   UBL will 'resume perfectly the height' when it is re-activated.    You probably don't want to hear the reason this is...  But you can activate and de-activate as much as you want and you will not confuse the printer about the 'correct height'.

studiodyne, 2018-02-11:
For information Ubl up infinitly and  go down infinitly just after  ,  after these situations

- If no print , just one movement to the outside , no bug
- If some g1 moves and go outside  in one command , bug !
- If a command sent to go from the bed , directly outside , the carriage stop on bed boundaries, and up and go down
- if go at the bed limit , and another command to cross the limit . No bug
- If disable g29 before go to pause from the center no bug

I think it's in the ' INTERPOLATION MATH ', if a  movement probed-zone is combined with a noprobed-zone the interpolation have breaks , and the G1 MOVE IS STOPPED on the Xbed and Ybed and doesn't make the complete travel

Try to find how it's possible to stop a G1 X500'nozzle park outside'   at  X430 'limit of bedsize '
It's not a true stop but the z axis need to raise a higher value , try to find , what happened a the exact moment of x=bedsize for example , it's at this moment something breaks

thks for help , i will use enable disable 

Steeve

Roxy-3D, 2018-02-11:
Agreed about the interpolation math.    The functions are coming up with a bad value if a NAN (invalid mesh point) is used in the calculations.   They should be coming up with 0.0 in that case.

studiodyne, 2018-02-11:
I trust you

studiodyne, 2018-02-11:
But , if you have a bed with a corner at 1mm
and you apply 0.0
The size of nozzle will shock the bed
It's why , you need to use the closest point to minimize the drama lol
Or create a define  like this  ' Outside Height Z' to have a fixed secure value for all outside movements 
lol

studiodyne, 2018-02-11:
UBL system force the user to activate an 'outside of bed ' secure height 
It's time to make it

studiodyne, 2018-02-11:
Good job ! lol

Roxy-3D, 2018-02-11:
Yes...  I understand why you are saying that.   But that is why it is very important that you setup and calibrate your printer.   If you tell the firmware the bed is a certain size....   Perhaps it would be good if the nozzle actually stayed in that area.    The problem with interpolating off of the nearest point (besides it taking a lot of calculations) is even that does not guarintee the nozzle won't be driven into the bed.

I'm more inclined to add an option something like:
```
#define UBL_Z_RAISE_WHEN_OFF MESH 5.00
```
If you really need to go off the mesh and UBL is active...   That number would be used as the correction instead of 0.000.      Would that be OK for your printer?

studiodyne, 2018-02-11:
I want to give you my secret ....
I have created a mad purge bucket ... magic ... and cost nothing ...
This bucket will give the possibility to never use prime tower for dual extrusion.
But purge bucket must be outside the bed and it's why i need a known behavior when go to the bucket 
But if you ensure that the interpolation maths permit to disable/enable ubl when go to the bucket it's enough for me .
And you have all the time to fix this in the future.
I will try your new solution tomorow
Thks for help 
Promise the secret of my purge bucket xill be revealed soon .

studiodyne, 2018-02-11:
Don't need new options 
Just add a comment on UBL description 

"To go outside of BEDSIZE enable/disable UBL" 

And ALL IS FIXED lol
And you have time to have reflexion about this and make some new maths

studiodyne, 2018-02-12:
After reflexion and for the happyness of everybody ...

> #define UBL_Z_RAISE_WHEN_OFF MESH 2.00

Is a very good idea
2mm is enough by default because feedrate of z never high and make time to raiseup

Lutz1967, 2018-02-13:
Hi! First I would like to thank you for the great work. I learn to use Marlin now since some month and it`s great to apply and create. Recently I`ve tried to make UBL work on RAMPS. Firstly I had used MESH_INSET=15 to utilize the inductive probe correctly, but then I got the Z-to-the-max issue and also Z into the bed. Then I read that the root cause is going outside the mesh and I`ve redone everything with  MESH_INSET=0 and corrected the mesh by hand. Then I made a test print and at one point (clearly within the defined mesh) the extruder pushed out all the filament ... I have no idea how extruder and UBL interacts but that is what happened. 
Any idea?

Roxy-3D, 2018-02-13:
I am currently working to resolve the Z-to-Infinity issue when the nozzle goes off the mesh.   

> Then I made a test print and at one point (clearly within the defined mesh) the extruder pushed out all the filament ... I have no idea how extruder and UBL interacts but that is what happened.
Any idea?

My guess is you have the filament sensor defined and there was some electrical noise on that signal.    If you are not really using the filament changer stuff, you might want to turn that off.

studiodyne, 2018-02-13:
Me too , every day , the use of marlin is a pleasure , babystep + ubl + fw retract + advance is my happyness

Lutz1967, 2018-02-13:
Thanks for the quick response. Look forward to use UBL when Z issue is fixed.

Filament runout stuff is not active. But I use the pin that is assigned to the sensor for something else (fan). I don’t think that should have any impact but I have corrected that.

Athemis, 2018-02-14:
@Roxy-3D probably slightly off-topic, but perhaps it may help to pinpoint the Z-to-Infinity issue: I observed that issue on my delta printer not only if the nozzle goes off the mesh during UBL levelling, but also during regular prints. Those instances all had in common that the model was very close to the bed boundaries set in Marlin, though the model never really exceeded those (at least the slicer did not complain). But you can trigger that reproducable by trying to print gcode that has been sliced using bed boundaries slightly larger than those set in Marlin. My guess would be that some kind of rounding error is involved here?

AnHardt, 2018-02-14:
Could yx-skew code be involved?

Roxy-3D, 2018-02-14:
>  probably slightly off-topic, but perhaps it may help to pinpoint the Z-to-Infinity issue: I observed that issue on my delta printer not only if the nozzle goes off the mesh during UBL levelling, but also during regular prints. 

I was working on that all day today.   It is not a rounding issue.  It is a problem with invalid mesh points being used in the Z-Height correction calculations.    And the problem with going off the mesh is there is not another set of mesh points to do the bi-linear interpolation.

studiodyne, 2018-02-14:
YEAHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHH

Roxy seems very closed to push something soon

Yes !!!!

Roxy-3D, 2018-02-14:
Right now I'm trying to fully understand the representation and usage of NAN in the Arduino library.   The reason is that is how the bad height is getting the stepper motors doing the wrong thing.   But something I don't understand is I have a configuration right now where doing a G26 causes the problem.  That is very strange (and others have reported this exact issue also) because G26 never sends the nozzle off the mesh.    

It will be a few days because I'm trying to fix this as cleanly as possible and not brute force the fix.  With that said...   I've got what I need now to get this resolved and I'm on it.

I'll probably do the Z_LIFT_WHEN_OFF_MESH as a separate Pull Request just because I have to change 37 Configuration.h files to do that one.

Roxy-3D, 2018-02-15:
@studiodyne    The requested changes are merged into bugfix_2.0.0.    I'm making a bugfix_1.1.x version now....

https://github.com/MarlinFirmware/Marlin/pull/9661

studiodyne, 2018-02-16:
YOU RULES !
I will test soon
Now , i'm working on the WANHAO LCD DLP , i just received it , at the software suit is so poor , two days and no print started
all is too young , and softwares are all prototypes , no smart function , for example , i want to print a tea cup , but to make economy of resin , i want to change the infill , with honey comb for example , and there is no software that transform a stl  with a hollowing/infill function ' and , the second question is ' how to get out the resin inside lolllllllllllll 
and more
Promise i test this soon

Roxy-3D, 2018-02-16:
My guess is if you enter Z-BabyStepping prior to doing the G26, it should work.   But really...   You want G26 to show you what the printer really sees.   The G26 pattern is not valuable if the user is moving the nozzle up and down.

filipgoc, 2018-02-16:
@Roxy-3D 
I understand your reasoning, which I believe is based on starting with the printer being able to print in the center before even launching UBL. However, my use case is very different from yours.

Long story short, I have a predictable and consistent tilt to the build platform. Thus, it is way easier for me, if not necessary, to have some leveling in place and activated to be able print anything - and thus to be able to babystep the proper z probe offset.  Only after that I would move into validating the mesh.

(If you're wondering why I don't have the Z offset form the nozzle set in config as precisely as possible, it's because I have quick swap extruders with various hotends. It's way easier to lower or raise z probe into the right neighborhood when I swap out a vastly different one, and then tweak the offset digitally in a few seconds.)

Does any of this make sense? 

I was poised to simply tweak the z offset as the validation pattern initiates (decidedly losing about a centimeter of validation) and then observe the rest of the pattern. Yet as I'm writing this, I suppose it would be indeed cleaner to add the z tweaking as an additional step prior to mesh validation...

I still wonder if it's necessary to _prevent_ user from using babystepping, if it actually is _by design._ If a user doesn't want to babystep, she or he doesn't have to. (User has to exhibit resolve, and act to babystep.)

Roxy-3D, 2018-02-17:
> I still wonder if it's necessary to prevent user from using babystepping, if it actually is by design. If a user doesn't want to babystep, she or he doesn't have to. (User has to exhibit resolve, and act to babystep.)

Baby Stepping is done by constantly calling idle() and having the LCD code constantly checking the encoder wheel's position.   G26 doesn't do anything to make that happen.   But it could.   It probably is as easy as calling lcd_status_screen() when long waits are happening.   (And testing the system for correct operation.)     If you want to wake up G26 to baby stepping...   I'll help you get that to happen.

filipgoc, 2018-02-17:
> If you want to wake up G26 to baby stepping...

Thanks. I want to keep you all focused on these bugs now :-) No distractions. 

I'll give it a bit more thought. Writing about this made me think that maybe what I want is adding a step to (my own) UBL flow.

studiodyne, 2018-02-17:
How i use UBL 
110°c bed , i make a cold mesh :smile , i save , i set fade heigh to 1mm , and it's all
To verify if UBL works , i have a tape on the z axis ,if the tape move like a flag , all is good
To verify ubl by lcd , i have made a custom user script ' G1 xy move ' because , the LCD move does not act the ubl
I don't use the g26 pattern it's too long , time is precious
One click on the babystep ' option autosave ' , and ITS ALL
And every day , i think about roxy that have created this AWESOME MAGIC FEATURE !!!!!!!!!!!!

studiodyne, 2018-02-17:
@Roxy-3D , please can you take one minute to read my post on your last pr conversation , i have made a little suggestion about ' Secure Z HEIGHT OFF MESh' , that works only when z axis is inferior to it , but when the z height have reached the secure value , you can disable this 
It's not good to have a z raise during all the print 

Example
```
if (end[Z_AXIS]>=UBL_Z_RAISE_WHEN_OFF_MESH) z_raise = UBL_Z_RAISE_WHEN_OFF_MESH;;
else  z_raise =0;

```

Studiodyne, 2018-02-17:
UBL_Z_RAISE_WHEN_OFF_MESH 2.5
Not implemented in 2.0 just the definition exist in config.h

Roxy-3D, 2018-02-17:
There was a problem with the use of #if ENABLED().    That had to be changed to #ifdef's because ENABLED() doesn't work with floating point numbers.     I'll try to bring up the bugfix_2.0.0 branch today and get it checked out (and working).

Studiodyne, 2018-02-17:
Don't work anymore on 1.1.x now

Roxy-3D, 2018-02-17:
Arrrrrgggghhhhhh!!!!

Studiodyne, 2018-02-17:
Not implemented too ?

Studiodyne, 2018-02-17:
I don't see the merged line ' calculations ' in the last bugfix 1.1.x

Studiodyne, 2018-02-17:
Travis not ok  loll

Roxy-3D, 2018-02-17:
@Studiodyne    I like the idea that above the UBL_Z_RAISE_WHEN_OFF MESH nothing happens.   And below that height, it simply raises to that height.   I think we can accommodate that.   But that will take a little bit more work because right now a lot of the work is being done in the Z-Height correction routines and they don't know the current height.

Studiodyne, 2018-02-17:
Take your time
All the release versions are very bad now
I keep my old release
Today , i have made a new up , and  i have lost ' big edit font ' , ubl still in progress , pause system is shit
Marlin not on the highlights for the moments.

Studiodyne, 2018-02-17:
My wish for UBL

    off mesh Z secure height 'you know all '
    off mesh z secure active only under current z
    lcd babystepping active is UBL disabled ' seems to be solved '
    And a new wish ..... FADE HEIGHT in LCD , because fade height change when the object to print change , and there is no possiblity to adjust it by lcd , and if we print by sdcard , to set a fade height need usb connection and will reboot the board
    Fade height have been forgotten , but it's a VERY IMPORTANT DATA
    I use one millimeter fade height , because i d'ont understand why10mm
    if you print a plate of 3mm I understand that we want same thickness everywhere , but it's the only object that need a high fade height , all other object want to smooth the plan the more quick as possible

Roxy-3D, 2018-02-17:
>  but it's the only object that need a high fade height , all other object want to smooth the plan the more quick as possible

No...  This isn't true.   The problem is the amount of extrusion isn't changed on a per layer basis.  The difference in the bed needs to be averaged into a large number of layers.

AnHardt, 2018-02-17:
Even fade height could be dynamic.
If it needs 10 mm (50 layers) to even out a 1mm failure, that is 1/50 mm/layer.
If the largest failure (max difference of the probed points) on the bed is only 0.1mm it only needs 1mm (5 layers) fade height, to have the same (1/50 mm/layer) MAX_HEIGHT_CORRECTION.

Roxy-3D, 2018-02-17:
> Even fade height could be dynamic.

Good point.   Fortunately...   This would be a relatively easy one to do within the SEGMENTED_MOVES logic because the firmware would know how much Z-Height correction was being applied in that area.

Studiodyne, 2018-02-17:
The idea of dynamic is so perfect  , but , because of the extrusion not compensed all is impossible
Fuck !!! the idea is exactly what we want , no setting all automatic !
The good news is ' Lcd fade height is implemented ' , but i never found it ..
I go to find where it is implemented and in what menu
Thks !

Studiodyne, 2018-02-17:
@Roxy-3D 
The LCD FADE HEIGHT not implemented in the UBL menu
I've seen the ultralcd.cpp
No implementation , on the UBL zone
Make a fix ,  add it on the list lol

Studiodyne, 2018-02-17:
Updated wishes 
1-) Lcd Fade height for UBL 
2-) Secure z height  ( + disable when current z raise to the same value )
3-) Fix big edit font lollllllllllllllll PLEASE my favorite functionnality !!! 
Thks in advance for the time you give here ! For us and for the firm

filipgoc, 2018-02-17:
For better or worse, this seems to be now temporarily a somewhat general UBL thread.

I am unclear about the use cases for **bed and hotend temperatures when probing the mesh.** 

- Is the ONLY reason to heat up the bed and the hotend before probing that most probes measure differently when heated up,  and the proximity to the hot bed (and extruder? really?)  is enough to mess them up? 
- If that's the case, then what we really want to do is  _match_ is the temperature when generating the mesh and the temperature when homing as a part of our start gcode. 
- If I 
    - home all axes first
    - and then heat up the bed
    - and then print, ...
    - then I should actually be generating a _cold bed mesh_ even if I'm printing at 235/70 °C? 

Is this right?

Roxy-3D, 2018-02-17:
Is the Bed Leveling main menu available when UBL is active?   If so, that is where it will be.

Roxy-3D, 2018-02-17:
> Is the ONLY reason to heat up the bed and the hotend before probing that most probes measure differently when heated up, and the proximity to the hot bed (and extruder? really?) is enough to mess them up?

Yes...   You will get different mesh values when probing cold or hot.   Personally...   I think one hot mesh is enough.   Because I use Z-BabyStepping when a print starts to manually adjust the first layer anyway.   But if you are not going to do that, you may want a mesh that is built at PLA temperatures and a mesh at ABS temperatures.   (And just save them to different mesh slots.)

filipgoc, 2018-02-17:
> I also don't really see a reason to generate a cold mesh. But it is in the menu system...

Hm. What do you think of this line of reasoning? : 

> what we really want to do is match is the temperature when generating the mesh and the temperature when homing as a part of our start gcode.

If I home the z axis while the bed is cold and then heat up before the print, shouldn't I actually be generating a cold bed mesh even if I'm printing at 235/70 °C?

Studiodyne, 2018-02-17:
@filipgoc On a 400x400 buildplate or a 600x600 builplate , it's strongly requiered to probe with heated bed , because all the mesh are really different , between 0.1/0.8mm , the hot bed mesh is not a joke , it's serious

@Roxy-3D  No available menu except the UBL big large strong infinite menu , so you can make a fix to have it

filipgoc, 2018-02-17:
@Studiodyne

I think I got it. 

It's _not_ as much about the probe changing the probing (as that should mostly just shift the mesh up and down), but about the bed actually changing shape due to heating up? Is that what you're saying? 

(Otherwise, like roxy, just shifting the mesh up and down as needed...)

Studiodyne, 2018-02-17:
yes it changes shape , and the most common shape is because , the bed have fixation on the left and the right , the heat increase size of the bed , and force is applied on the fixation ,  the curve of the bed change , it's very light for eyes , but for a robotic nozzle absolutely perfect in an xy plan , you can be sure , in the center you have the nozzle that can not extrude , and in the corner  the filament not glue to the bed
It's why , you have to make the better probing .

Studiodyne, 2018-02-17:
it's why i just use 3 points matrix for ubl ,    one left , one center , one right , same for y axis , and it's enough , i don't have 32 bits computer so it's enough
and works REALLY PERFECT

Studiodyne, 2018-02-17:
The curse of the perfect first layer , is madness , it's why  i have my own limits !! lol

filipgoc, 2018-02-17:
Right. I think for me it has not been as markant, as I don't have any machine with 400mm+ bed size and I use relatively thick aluminium (3+mm)... 

But of COURSE it makes sense. I was blindsided by thinking about the probe to the point that I forgot about the bed. 

Yet, this leaves the question of hotend.
- Why would we heat up the hotend? Because we presume proximity to the probe that would affect it (which is clearly very presumptuous)? Or because we are getting ready for G26? (In which case it'd be better to heat up later so that we are not baking an empty chamber.)

Roxy-3D, 2018-02-17:
The hot ends expand when heated.   You can position your nozzle .05 mm above the unheated bed...   And when you turn on the nozzle heat, it is very possible you will see the nozzle touch the glass.

Studiodyne, 2018-02-18:
@Filipgoc ,  in robotic 0.1mm is STRONG , so ,  all the closest conditions of printing must be reproduced for probing , and , we are all agree that it's really not difficult to heat nozzles and bed , for probing , it's why , we make it , and it's sure we will have a really better and more precise mesh
And no really a problem , because , i have made my mesh , a few month ago , and i still use it  today ,  the mesh is not deleted when you upload a new firmware

UBL is the autolevel that can have a medal of honor because EXTRAORDINARY AND EASY !
And this is  The @Roxy-3D  creation
Girlpower !

Roxy-3D, 2018-02-18:
> 2-) Nan/offmesh UBL bug

I believe this one is gone.

Studiodyne, 2018-02-18:
Coool 👍 

I test it soon , it's easy for me to test , i have a dual extrusion print , with purging in an offmesh bucket , every tool change ,  you imagine 250 tests in one print lol

filipgoc, 2018-02-18:
@Studiodyne @Roxy-3D Thanks for helping clarify the variables and the context of the heating or no-heating for probing!

The sequencing is worth thinking about.
99% of time I'm printing PETG or PLA these days. 
I'l looking for a compromise then: 

I think what I'm gonna do in IN THE NAME OF UBL, all hail UBL, is this: 

Generate the mesh at 170/60. 
```
M104 S170   ; Hot enough so that leftover filament is soft and squished if necessary,
but no so much that it's oozing on my lovely PEI sheet. Mesh precision wise,
this doesn't matter because of babystepping. At this point we don't care. 
M190 S60    ; I don't normally print ABSes or polycarbonates, so this will
not be crazily off for by far most of my prints. 
```

Then, babystep at the very top of the range for PLA. 
```
M109 S225
; print something and adjust the z height to perfection
```

And combine with such start gcode
```
G28         ; home
(G0 X_ Y_    ; get the nozzle off-platform. in my case, safe homing does that already.)
M109 S170   ; heat up to the mesh temp, also nice to wait at lower temp while the bed
is heating so we're not baking the nozzle
M140 S60    ; heat up to the mesh temp (not sure if M190 is necessary here. thinking not)
G28 Z       ; Re-home Z for an updated value
M109 S{print temp} ; finish the heat up
M190 R{print temp} ; finish the heat up
; here, wipe nozzle if you have a nozzle wiper
; print
M117 Print starting
```

Roxy-3D, 2018-02-18:
And I know people are tired of hearing it...    But Turning on the DOUBLECLICK_FOR_Z_BABYSTEPPING makes it much easier to get a perfect first layer independent of temperature.   As the printer is drawing the skirt,  You just adjust the nozzle so you are getting the right amount of 'squish' to line being put down.  

Probably it makes sense to have a perfectly calibrated machine for the filament you usually use.   But when you switch to another filament type for a print...   DOUBLECLICK_FOR_Z_BABYSTEPPING will make it so that print turns out OK.

[35 more replies]

closed 2018-02-28 by Studiodyne